ArduPilot/ardupilot_wiki #3: copter menu points to missing pages

rmackay9, 2015-01-09:
the optional hardware menu includes a link to relay: http://copter.ardupilot.com/wiki/common-relay/

but this in turn directs to this other page: http://copter.ardupilot.com/wiki/common-relay-2/

Craig3DRobotics, 2015-01-09:
Note the comment

Please see common-Relay Switch
<http://copter.ardupilot.com/wiki/common-relay-2/>.

This is an orphan page, created when the common page on planner was
redirected. It may be deleted when all links to this page have been updated
to point to the new page (above) and this page has been removed from any
menus.

The page is a WIP / placeholder / reminder while Hamish re-works the menus.

There are a bunch of them just like it.  Since Hamish is aware of them I'm
not sure there is much point in reporting them all here.

hamishwillee, 2015-01-22:
There isn't. I believe I have now fixed the Rover wiki. Working on Copter and Plane today/tomorrow. When I think I am finished then if I should happen to miss any that would be a good opportunity to post any you find :-0

hamishwillee, 2015-01-23:
I believe this is fixed. I have fixed up all the pages so that the original pages (mostly) remain in the menu with the current information, and all the duplicates have been permanently deleted. It is possible I have missed a case or two, but given that the menus on the site all appear intact after the big deletion, that is relatively unlikely.

closed 2015-01-23 by hamishwillee