ArduPilot/ardupilot_wiki #4: Sik Documentation points to the wrong wiki

Craig3DRobotics, 2015-01-14:
https://github.com/tridge/SiK
points to http://code.google.com/p/ardupilot-mega/wiki/3DRadio
but that site has the technical details of the radios that are missing from the new wiki
we need to add the technical details to the dev wiki and rationalize the other radio pages

hamishwillee, 2015-03-16:
> but that site has the technical details of the radios that are missing from the new wiki
> we need to add the technical details to the dev wiki and rationalize the other radio pages

This has been done as #9 (I just checked the (now deleted) page /ardupilot-mega/wiki/3DRadio and everything has been captured in the upated version in:
http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-using-the-3dr-radio-for-telemetry-with-apm-and-px4/

So "someone" needs to update https://github.com/tridge/SiK to point at the above link. I'll leave that with you Craig :-)

gmorph, 2015-03-17:
That link doesn't work.  Did you mean this?

http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-version-2/

Let me know the right link and I'll get Tridge's github page updated.

Thanks

Grant

hamishwillee, 2015-03-17:
Hi Grant, Yes, that is the correct link: http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-version-2/

(or just http://planner.ardupilot.com/common-3dr-radio-version-2/ works too.)

CraigElder, 2015-03-17:
Updated here

https://github.com/Dronecode/SiK/commit/1c545906fffde40855b756f1ad122f852633b7da

pull request here:

https://github.com/tridge/SiK/pull/23

hamishwillee, 2015-03-17:
Thanks. I'll close this then.

closed 2015-03-17 by hamishwillee