ArduPilot/ardupilot_wiki #11: Mission Planner: Decouple MP/GCS and device setup walkthroughs

hamishwillee, 2015-01-21:
There should also be some focus on decoupling the documentation of showing Mission Planner walk throughs on how to configure the device. At best the ardupilot docs should discuss in principal what needs to be configured and the walkthroughs should be a link to numerous GCS pages for the setup walkthrough. This is important as we also have tablet devices that also do the same job as the desktop GCSs.

(BILL B)

auturgy, 2017-10-15:
This is still a relevant point imho, but it's a lot of work to provide setup guides for every GCS tool.  Thoughts?

hamishwillee, 2017-10-15:
I agree. When I wrote this I owned this work, and this was a speculative todo list.

auturgy, 2017-10-16:
MAVproxy has its own docs, as does APMPlanner2.  The QGC docs (https://docs.qgroundcontrol.com/en/SetupView/SetupView.html) cover both PX4 and ArduPilot, so would better linking to the QGC, MAVProxy and APMPlanner2 docs achieve the key outcome?

hamishwillee, 2017-10-16:
Not sure. I didn't start the work so I didn't do much experimentation.

brunoolivieri, 2019-07-12:
Would this issue still make sense to stay open?

auturgy, 2019-07-16:
In my opinion we should make it easier to find instructions for the popular GCS packages, particularly QGC as it is the only actively supported free option for handheld devices at the moment.  I don't think we need to keep this issue open though.

closed 2019-07-16 by auturgy