ArduPilot/ardupilot_wiki #14: Tutorial videos coming up as private (plane)

hamishwillee, 2015-01-21:
All but two of these videos are coming up as private on YouTube: http://plane.ardupilot.com/wiki/videoguides-3/

(originally posted here: http://ardupilot.com/forum/viewtopic.php?f=51&t=4704 )

hamishwillee, 2015-09-09:
Closing. The private videos cannot be made public because no one owns them. I have therefore deleted them and replaced with a better set.

closed 2015-09-09 by hamishwillee

sediq-khan, 2019-02-11:
Hi,
The link to the videos is not available, could you please let me know where can I find the Videos. On a separate note, I am adding a new message to the MAVLink list of messages, and want to display it on the Mission Planner,  I would appreciate your insight/advice on how to add the message and how to display it on the mission planner.

Thank you