ArduPilot/ardupilot_wiki #8: CHDK has duplicate docs

hamishwillee, 2015-01-21:
You can see duplicates in search: http://copter.ardupilot.com/?s=chdk&submit=Search

hamishwillee, 2015-01-21:
This was also posted in the forum here: http://ardupilot.com/forum/viewtopic.php?f=51&t=6227#p11365 (has additional detail)

hamishwillee, 2015-02-03:
When  rationalize the CHDK pages check useful drawings on this page:  http://diydrones.com/forum/topics/using-aux-pins-as-relays-for-chdk?commentId=705844%3AComment%3A1898413&xg_source=msg_com_forum

hamishwillee, 2015-03-19:
OK, so I have looked at the articles that show up as a result of [search](http://copter.ardupilot.com/?s=chdk&submit=Search). There are only two topics on CHDK: [APM to CHDK Camera Link Tutorial](http://planner.ardupilot.com/wiki/common-apm-to-chdk-camera-link-tutorial/) and [CHDK Camera Control Tutorial](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-cameras-and-gimbals/common-chdk-camera-control-tutorial/).

The second is much more useful. The first uses shows one implementation with a hacked cable and using a relay. However it doesn't really explain *how* the CHDK works/integrates with APM so it would be hard to extend this to do more. By contrast the second article provides a full explanation of what it is trying to achieve and how it does so.

The second article has some duplication and needs to better cover Pixhawk.  I propose we delete the first article because it isn't particularly extensible.

gmorph, 2015-03-22:
I agree the second article is more useful but it doesn't describe how to
create that cable.  Any chance we can put the cable creation info from the
first page onto the second page then delete the first?

Thanks, Grant.

hamishwillee, 2015-03-22:
Craig felt the same way. More importantly, on further reading I came to understand that this uses a Relay rather than servo, and that is a slightly different story. I will try to merge the two, but otherwise I'll rename the first one to differentiate it better.

hamishwillee, 2015-03-26:
I completely withdraw my original suggestion!

If all you want to do is trigger the camera shutter (which is basically the only level of camera integration in ArduPilot) then the approach in the second article is much simpler. On APM you just connect the output to a relay. On Pixhawk I believe you may also need to add a voltage booster to get the signal high enough to trigger the USB. Not sure if that applies to relays/servos/both but I suspect may work "out of the box" for relays.

A PWM setup here will work too, but you need a board to convert the PWM to a signal.

The second article looks great on the surface, but actually doesn't integrate properly with the camera shutter trigger code in ArduPilot. This means that if you're running a survey mission you'll have to manually add the appropriate shutter commands as "DO_SET_SERVO". The benefit is you can send multiple commands -e.g. to set zoom etc.

I need to find out if, once you have already set it as a camera trigger, you can send different PWM signals down the servo output - perhaps using DO_SET_SERVO.

hamishwillee, 2015-03-30:
Both articles still exist. However as they serve different purposes I've positioned them so it is more clear which you should use in different circumstances. I've removed the duplication so that mission planning and configuration of the output in ArduPilot is in separated (where appropriate). These are still waiting review, but I am setting as closed for now.

closed 2015-03-30 by hamishwillee