ArduPilot/ardupilot_wiki #9: common-Using the 3DR Radio for Telemetry with APM and PX4

hamishwillee, 2015-01-21:
The documents for the different radios contain a lot of duplication. The proposal is to split them under a landing page with the 3Dr version 2 and version 1 documents containing only the hardware specific information. The country specific information should also be separated.

Craig3DRobotics, 2015-01-22:
I would say separate them so that the normal user setup and usage is
explained in one part, the AT commands and country specific details etc are
in a separate Advanced section.

hamishwillee, 2015-03-13:
Started job by creating [common-Radio Standards and Regulations](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-radio-standards-and-regulations/) from the (merged) "Support for different countries/regions" sections in docs:

* [Using the 3DR Telemetry Radio with APM 2.0](http://plane.ardupilot.com/wiki/3dradio-3/#support_for_different_countriesregions) (plane)
* [Using the 3DR Radio for Telemetry with APM and PX4](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-using-the-3dr-radio-for-telemetry-with-apm-and-px4/#support_for_different_countriesregions) (common)
* [3DR Radio version 2](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-version-2/#support_for_different_countriesregions) (common)

The pages above retain a stub section pointing to the new docs. Everything is linked and in menus

hamishwillee, 2015-03-13:
PS, Craig, I will have a look at the docs today and take your suggestion under due consideration :-)

hamishwillee, 2015-03-13:
From Craig FMI - Only differences between 3DR Radio v1 and v2 are:

* form factor
* Physical connections - One needs FTDI cable, the other has it built in
* No 433 version of rev 1

hamishwillee, 2015-03-15:
Deleted the plane doc as it is a duplicate http://plane.ardupilot.com/wiki/3dradio-3/
The section on support for different languages was in advance of the other docs, but this has now been merged.

hamishwillee, 2015-03-15:
Hi Craig

After much thought I've decided to merge the docs into one, and attempt to section them as you've indicated. 

While it would be nice to have separate docs for the two radios, there would be quite a lot of duplication as they are so similar. Given how similar they are, I think one well-sectioned document is the way to go. 

We have two docs:
* 3DR Radio version 2 (common)
  * Covers just v2 radio. 
  * Covers the "have to know" stuff - wiring, configuring using mission planner. Using with tablets etc.
* Using the 3DR Radio for Telemetry with APM and PX4 (common)
  * Covers all of above, but also what each of the settings allows you to do.

So the first is very much in the "user guide" style and the second is "users and advanced users". There is lots of duplication.

Main options:
1. Make the

hamishwillee, 2015-03-16:
First draft complete. Requesting feedback from ardu-wiki editors. http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-using-the-3dr-radio-for-telemetry-with-apm-and-px4/

Have not deleted the other docs.

CraigElder, 2015-03-16:
I just read the new page and it looks great.

hamishwillee, 2015-03-17:
Now split as:

* [3DR Radio v2](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-version-2/) -user guide and overview
* [3DR Radio — Advanced Configuration](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-advanced-configuration-and-technical-information/) - all the extra configuration stuff
* [3DR Radio v1](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-v1/) - just the connection information for v1

All other articles deleted. A few broken links may appear but I will fix these as we come to them.

Much better. I am marking this as closed.

closed 2015-03-17 by hamishwillee

gmorph, 2015-03-17:
They look great Hamish - thanks!