ArduPilot/ardupilot_wiki #15: Remove CLI from the wiki

hamishwillee, 2015-01-22:
The CLI is being phased out on all platforms. The features were massively cut down with AC3.1.5 and then it was removed for the APM2 completely anyway with AC3.2.  All CLI functions are now available via regular MAVlink commands in the GCSs.

Consensus is to delete all mention. I'd personally be happy with at least a glossary mention.

These are the pages that need to be removed from the copter wiki:

* Instructions > Advanced Configuration > Command Line Interface (CLI) : http://copter.ardupilot.com/wiki/using-the-command-line-interpreter-to-configure-apmcopter/
* TOC > Advanced Configuration & Tuning > Using the Command Line Interface: http://copter.ardupilot.com/wiki/using-the-command-line-interpreter-to-configure-apmcopter/
* TOC > Advanced Configuration & Tuning > Using the Command Line Interface > CLI Setup Test Modes : http://copter.ardupilot.com/wiki/common-cli-setup-test-modes/
* TOC > Advanced Configuration & Tuning > Using the Command Line Interface > RC Input : http://copter.ardupilot.com/wiki/common-rc-input/
* TOC > Advanced Configuration & Tuning > Using the Command Line Interface > EEProm : http://copter.ardupilot.com/wiki/common-eeprom/

http://copter.ardupilot.com/wiki/common-sensor-testing/
MOVE THIS ONE TO TRASH: http://copter.ardupilot.com/wiki/using-the-command-line-interpreter-to-configure-apmcopter/  ß moved to trash
 
The pages should of course be deleted from common wiki, and anywhere else they appear.

Craig3DRobotics, 2015-01-22:
Let's put the CLI information into the dev section in the documentation.
There are some commands and some details about the NSH shell that would be
good to keep around.

hamishwillee, 2015-01-22:
Craig, any particular documents from the above to keep, or all? (or read carefully and use my judgement?).

Note, this will take a while as I am trying to simultaneously debug and avoid breaking the menus.

hamishwillee, 2015-01-23:
Following discussions with Craig and Randy I have removed all the CLI content from the menus in planner, copter, plane, rover. I have deleted some content and moved other to dev. I will outline "next steps" in next comment. Detail below:

* [common-eeprom](http://copter.ardupilot.com/wiki/common-eeprom/) deleted completely from all sites (does not work, and not a particularly useful "use case")
* [common-rc-input](http://copter.ardupilot.com/wiki/common-rc-input/) deleted completely from all sites (functionality no longer present)

* commonmission-planner-command-line-interface-cli/ deleted from plane/copter/rover/planner 
  * Moved to [Dev here](http://dev.ardupilot.com/wiki/commonmission-planner-command-line-interface-cli/) 
  * Added a note to top explaining CLI is deprecated

* using-the-command-line-interpreter-to-configure-apmcopter/ - deleted from copter
  * moved to [Dev here](http://dev.ardupilot.com/wiki/using-the-command-line-interpreter-to-configure-apmcopter/) 
  * Added a note to top explaining CLI is deprecated

* Updated [common-cli-setup-test-modes](http://planner.ardupilot.com/wiki/common-cli-setup-test-modes/) with current running tests. This still needs to be moved to Dev (following review)

* [using-the-command-line-interface](http://dev.ardupilot.com/wiki/using-the-command-line-interface/) : this is a parent doc that was already on dev. I Added a note to top explaining CLI is deprecated

As an aside I also removed just the first video in http://planner.ardupilot.com/wiki/common-sensor-testing/ on "Raw sensor display: Enable Raw Sensor Values and Display Sensor Data on Graph Video How To’s Below." This is otherwise fine to leave in planner. 

Status
--------

These are now on Dev and need to be merged
* Moved to [commonmission-planner-command-line-interface-cli](http://dev.ardupilot.com/wiki/commonmission-planner-command-line-interface-cli/) 
*  [using-the-command-line-interpreter-to-configure-apmcopter/](http://dev.ardupilot.com/wiki/using-the-command-line-interpreter-to-configure-apmcopter/) 
* [common-cli-setup-test-modes](http://planner.ardupilot.com/wiki/common-cli-setup-test-modes/) with current running tests. *This still needs to be moved to Dev (following review)*
* [using-the-command-line-interface](http://dev.ardupilot.com/wiki/using-the-command-line-interface/) : this is a parent doc that was already on dev.It is probably where the merge should occur.

Some of these are now waiting approval in ardupilot

hamishwillee, 2015-01-23:
I will also run search and clean remaining mentions:
* http://plane.ardupilot.com/?s=CLI&submit=Search
* http://copter.ardupilot.com/?s=CLI&submit=Search
* http://planner.ardupilot.com/?s=CLI&submit=Search
* http://rover.ardupilot.com/?s=CLI&submit=Search

hamishwillee, 2015-01-24:
Removed CLI mentions from following without seeking review:

* http://planner.ardupilot.com/wiki/common-simulation/ 
*  http://planner.ardupilot.com/wiki/mission-planner-toc-planning/

hamishwillee, 2015-01-24:
Have to move http://planner.ardupilot.com/wiki/mission-planner-terminal/ to DEV and it becomes the root doc in DEV. Top image with "Advanced view" circled is important.

Check for Terminal everywhere - CLI.

closed 2015-01-26 by hamishwillee