ArduPilot/ardupilot_wiki #16: Parent menu items that are deleted permanently break main menu

hamishwillee, 2015-01-23:
If a menu item with sub menu items is deleted, then the sub items are displayed at the wrong level, causing disruption of the main menus. This is a.k.a. "Phantom tortoise problem".

We need either:
1. Mechanism to prevent items being deleted (with reason) if they are in menus
2. Mechanism to ensure that menus still render properly even if parent item is deleted
3. Ideally a good way to detect if items in trash or delete permanently are in menus.

hamishwillee, 2015-01-26:
A bit more on the behaviour, as discovered by testing ....

Item is sent to trash
* User logged in: item and sub items displayed in menu but links are broken.
* User logged out: item is not displayed in menu. If the item has sub items, these are not displayed in the menu either.
* Menu admin retains all menu items - nothing visible to show item removed

Item is restored from trash
* Original menu restored

Item is permanently deleted
* Any sub items become top level items
  * On top level menu this results in phantom tortoise issue
  * On sidebar these are not being displayed. Mostly likely because not below the root menu.

hamishwillee, 2015-01-26:
There does not appear to be a mechanism in menu (e.g. CSS) to highlight items that need to be deleted. I think best option is to have a plugin that iterates all the menus and prevents deletion if current item present. Need to discuss when putting time into doing this is reasonable (ie priority).

hamishwillee, 2015-09-09:
All pages are now in the main menu, so removing before deleting should be part of the process. Still worth thinking about plugin to address this case.

gmorph, 2016-11-15:
Hamish - we have migrated to a new Wiki system so I'm not sure if this issue is still relevant.  I'm going to close this out but if you think it should remain open just re-open it.  If you do re-open it I think we need a description of how the new wiki system might lose parent links.

closed 2016-11-15 by gmorph

hamishwillee, 2016-11-16:
@gmorph This was about the old system. While the same thing would happen in the new system, there are better warnings in the debug output, so closing this "right thing to do".