ArduPilot/ardupilot_wiki #21: Lidar-Lite wiki: we should add info re the cap that goes between VCC and GND

rmackay9, 2015-01-26:
There's a discussion on diydrones about the exact cap that should be used.  I think Philip, Chris Anderson and Tridge all know which cap should be used.

rmackay9, 2015-01-26:
Note, there's a discussion here on diydrones where Sandy asks for this.  
http://diydrones.com/profiles/blogs/cheap-ir-lidar-lite-90

hamishwillee, 2015-01-26:
Page updated with new image.

rmackay9, 2015-01-27:
hmm.. I don't see the cap in that diagram...

rmackay9, 2015-01-27:
Ah, sorry, that was because I hadn't approved it.  done! thanks!

closed 2015-01-27 by rmackay9

zlite, 2015-01-27:
I've created a schematic and PCB design for people who want to make their
own low-pass filter boards. It's here
<http://123d.circuits.io/circuits/514802-lidar-lite>.

hamishwillee, 2015-01-27:
Hi Chris

Do you think it is worth adding this to the page?

For example replace the text:
"*Note: The capacitors shown in the diagram are required for first
generation hardware (as available Jan 2015). These should not be required
in later versions of the LIDAR Lite (at which point the diagram will change
to this
<http://planner.ardupilot.com/wp-content/uploads/sites/5/2014/08/RangeFinder_LIDARLite_Pixhawk_v2.png>
and
will have this circuit diagram
<http://planner.ardupilot.com/wp-content/uploads/sites/5/2014/08/LIDAR-LitePixhawkConnectionCircuitDiagram.png>.*
"
with
"*Note: The low pass filter shown in the diagram is required for first
generation hardware (as available Jan 2015) - a schematic and PCB design
are provided here <http://123d.circuits.io/circuits/514802-lidar-lite>.
These should not be required in later versions of the LIDAR Lite (at which
point the diagram will change to this
<http://planner.ardupilot.com/wp-content/uploads/sites/5/2014/08/RangeFinder_LIDARLite_Pixhawk_v2.png>
and
will have this circuit diagram
<http://planner.ardupilot.com/wp-content/uploads/sites/5/2014/08/LIDAR-LitePixhawkConnectionCircuitDiagram.png>.*
"

in
http://planner.ardupilot.com/wiki/common-rangefinder-lidarlite/#Connecting_to_the_Pixhawk

Regards
H

zlite, 2015-01-27:
There's probably no harm in adding it as you suggest now. Later on, once
the module has been updated with these filters built-in, we can delete it.

Thanks!

Chris

hamishwillee, 2015-01-27:
Thanks! Done (pushed live - no review needed)