ArduPilot/ardupilot_wiki #24: Tidy CLI pages in Dev Wiki

hamishwillee, 2015-01-27:
All CLI content was moved to dev wiki as part of #15 . This task is to tidy those remaining documents. 

    Mission Planner Terminal
        common-CLI Setup Test modes
        common-Mission Planner Command Line Interface (CLI)
        Mission Planner Command Line Interface (CLI)
        Using the Command Line Interface
        Setting up flight modes with the CLI
        Using the Command Line Interface to Configure APM:Copter

auturgy, 2017-10-15:
@peterbarker is the CLI dead yet?  Can the pages be removed and this issue closed?

peterbarker, 2017-10-16:
On Sun, 15 Oct 2017, auturgy wrote:

> @peterbarker is the CLI dead yet? Can the pages be removed and this issue
> closed?

CLI is dead.

auturgy, 2017-10-16:
> thread [2].
> Links:

thanks: I'll dig through and kill the CLI pages tomorrow.

Hwurzburg, 2019-10-10:
@auturgy can this be closed now?

Hwurzburg, 2019-11-02:
all pages are archived...closing this

closed 2019-11-02 by Hwurzburg