ArduPilot/ardupilot_wiki #32: Transposh plugin turned off

hamishwillee, 2015-02-02:
Transposh support was turned off during debugging. This may have been the cause of some instability, and is being investigated.

bys1123, 2015-02-06:
Hello, I'm the leader of unofficial chinese APM wiki site, ardupilot.cn
We've translated about a hundred page now, that helps lots of APM user in china, APM is very famous at drone/RC forums in our country now.
We are using a popular online translation tool named Transifex for team work from two weeks ago.
https://www.transifex.com/projects/p/apm/resources/
It's friendly for open source, it's free for this project, easy to use, and have lots of API and plugins.

So, maybe the official wiki could find a way use Transifex for some translation teams like us, we can contributing to this better.
e.g. Add wiki raw content to github. It will easy to add page to Transifex, and this page will auto update on it, we will know that page needs to renew very fast.
Machine translate to chinese only have about 30% could be understanding. Human translation is far beyond that.
There is lots of chinese people truly loves open source project now.
We're glad to join in the official site but it's very slow to visit for chinese users.

hamishwillee, 2015-02-08:
Hi 斯东Stone, I will investigate and get back to you. I will also respond in more detail to your message on the ardu-wiki editors group. THanks for your post!

hamishwillee, 2015-02-09:
Transfirex looks like a good system. What is the benefit of writing to Github over writing direct to Transfirex? How does transfirex live compare - superficial research seems to indicate that it will make tracking change very easy, and mean there is no need to do anything much on the site for getting the content. 

If we chose we could use https://wordpress.org/plugins/transifex-live/ plugin to also display this content.

Transposh is a well respected plugin. It is among a number of services that offer automated translation with replacement by custom translation. Some of the paid services have a slightly better reputation. The disadvantage of this type of service is that the automated translations are often bad, and that the translated pages are not stored in the database and are hence not searchable. 

Alternatives include plugins like PolyLang, where translations are stored as separate pages in the database. These are searchable and always known to be good at time of translation. Weakness appears to be in keeping this "in-sync".  Transposh and Transfirex both look pretty good from this perspective. 

One other alternative I see is a hybrid combination of Transfirex and PolyLang. In this approach we could use Transfirex for translations, but keep all translations and originals synchronised in Github. Then we could use a script to sync github to the Polylang copies. This would allow local searching as well. A lot more effort though. 

More research required to establish how useful automated translation is. 

Some research:
* [How To Make A Multilingual WordPress Site: Best Translation Plugins](http://wplang.org/translation-plugins-languages/#5) (2015)
* [Best WordPress Translation Plugins](http://www.wpmayor.com/best-wordpress-translation-plugins/) (2012)
* [How To Translate Your WordPress Website Into Multiple Languages] (2014) (http://www.wpexplorer.com/translate-wordpress-multiple-languages/)

bys1123, 2015-02-09:
Thank you for these information. Very helpful.
But this is a very serious problem. Our Internet is in a "Great firewall", we can not load google fonts and Gravatar in Wordpress dashboard, so every page is blank on load we should wait these blocked resoruce timeout then the page will show up, it needs about 1 minute. 
So, our wordpress needs to delete these code. 

For wiki page, it needs 10s in north, 30s in sourth. Chinese users use famous website most of time, it's just need 2~3s. If it's very slow, people will not use it anymore.
Mission Planner is slow in sourthern china, we've added the code that use a mirror firmware server to speed it up.

Can you edit the code of PolyLang, if select chinese, this page will jump to chinese site, it just need replace .com to .cn in URL.I promise we will keep every link in wiki.

hamishwillee, 2015-02-09:
Thanks for the clarification. You've convinced me that we won't be able to help you through any hosting this side of the great firewall.

> Can you edit the code of PolyLang, if select chinese, this page will jump to chinese site, it just need replace .com to .cn in URL. Chinese site would reverse this, jump back to selected language page on ardupilot.com, and when we translate, we will keep every link in wiki, don't worry about that.

Whatever solution we use has to work for all languages. I will certainly keep this request in mind as we design a solution.

hamishwillee, 2015-02-10:
The proposal - under discussion.

We will write a script to dump the wiki source(s) to github. This will include a no-translate section at the top of the page with article metadata - categories, slugs, tags, parents, site etc. This satisfies [Stone's](https://github.com/bys1123) request. Plan to do this in the next calendar month.

For the "general wiki translation case" we will *investigate* moving from Transposh to a system that uses PolyLang, Transifex and Github. 
The solution would look like:
*  Translated topics managed by PolyLang
* Script to copy untranslated content to Github (as above) and translated content from Github into translated topics
* Server to synchronise Github and Transifex content.

The translations for all languages would then all be made in Transifex (we would prevent local editing of translations) using their great translation tools. On the site we would enable searching of the different languages. If however we decided to change our solution from Transifex the local copy on wiki would remain undamaged, and we could arrange some sync to the new system from Github.

How much of this gets done depends on other priorities, but moving to PolyLang is likely.

hamishwillee, 2015-03-12:
Have created prototype of dump output here: https://github.com/hamishwillee/ardupilot_wordpress_sources - needs checking to see if format appropriate, and may need to host elsewhere. Under discussion in wiki editors group

CraigElder, 2015-03-13:
Per Hamish's request the repo is now located at
https://github.com/diydrones/ardupilot_wordpress_sources

hamishwillee, 2015-04-27:
I reviewed Polylang, Transifex and other plugins to determine our best path forward for localisation. In at least the medium term I have re-enabled and updated Transposh,

The plugin now has the following modifications/settings:
* Ran the tool to "remove errors from older versions". This removed errors in the network logs on loading pages. There are still some errors in the apache log, but these are not frequent (as previously) and may predate running that script.
* Remove automatic translations. Query from users indicates that these are deceptive. From now only only "real" translations will be supported.
* Added sidebar support for Spanish, Korean, French, German - most popular languages by hits on our site. Chinese not enabled (though popular) as there is a local site for that.
* Remove access for anonymous users to post translations. Translations can be posted if a login has been created for the user (subscriber) though they will not automatically have edit rights of the original content. There is no approval process for translated content.
* Allow search within translated content. (to be reviewed - does not appear to throw up false positives in English search, but this needs to be checked).

Transposh supports most of what we want and need:
* It provides in-place translation, that can detect when a string has been modified or added (ie change is visible and clear)
* Access can be given without providing access to the original content (low risk)
* It works out of the box
* We can disable automatic translation.

What it does not do is:
* Provide an easy way to note that something has changed and needs retranslating (unless you are already in the article)
* Provide any easy way to export the data to use other/external tools.

I reviewed Polylang, with the idea of creating a mechanism whereby we would export content to github, translate in transifex and then round-trip import back into the wiki. This would give complete flexibility on our translation solution, great tools etc. I decided against because:
* Much of the infrastructure is not in place and would have to be invented - more complicated and time consuming to set up
* Support to get menus right (in particular) would be complicated - we would need to manage on all the wikis which would be complicated.

hamishwillee, 2015-04-28:
Have added automatic update of the [copter/planner source to github](https://github.com/hamishwillee/ardupilot_wordpress_sources ) as discussed. This should update every week, Sunday at 8.05 am. 

I believe this issue is now closed.

closed 2015-04-28 by hamishwillee