ArduPilot/ardupilot_wiki #35: varnish cache is not being auto-invalidated on page update

tridge, 2015-02-03:
When we edit a wiki page the new page should be immediately visible to all users. To support that we use a wordpress plugin to tell the varnish cache to invalidate any cached content for that page. That used to work, but stopped working on one of the wordpress updates. We need to diagnose and fix that problem.

hamishwillee, 2015-02-03:
Hi Tridge,

I just edited http://planner.ardupilot.com/wiki/table-of-contents/common-documentation-page-structure-test/ on, then reloaded the page in another browser (not logged in). The information seemed to have updated correctly. 

*How are you reproducing this?*

We're using https://wordpress.org/plugins/better-wp-varnish/ which asserts to update posts, but which is only supported to WP3.8.3. 

W3 Total Cache currently does not have the IP of the varnish server specified in its [reverse proxy settings](http://ardupilot.com/wp-admin/network/admin.php?page=w3tc_general) so we could try turning this on too.

I would expect W3TC above to work if we enabled it, but other options include:

* https://wordpress.org/plugins/varnish-http-purge/
* https://wordpress.org/plugins/multi-varnish-http-purge/
* https://wordpress.org/plugins/wordpress-varnish/
* https://wordpress.org/plugins/varnish-purger/

tridge, 2015-02-03:
You have to make sure the page is in the varnish cache before the test. So if I load a page in a non-logged-in browser such as this page:
http://plane.ardupilot.com/wiki/flying/automatic-landing/
then make a small edit, then force reload in the non-logged in browser I don't see the change.
It is possible the problem is intermittent. I just tested on the above page and the problem did happen.

hamishwillee, 2015-02-03:
Just a quick check - is it correct now? 
I loaded the above page in a fresh browser and it was correct (assuming the change was "mininum" to "minimum").
If you fixed it, can you let me know next time you find an example.

hamishwillee, 2015-02-03:
OK, agreed. I can reproduce. I'll see if updating total cache fixes this.

hamishwillee, 2015-02-03:
Tridge,

So if I modify a page on Chrome then I see the changes on my Firefox (not logged in) immediately. If I reload the same page that I have previously opened in a VM (Ubuntu) firefox it is not updated. However if I restart varnish then it is. I don't really understand why the Firefox behaves differently on local machine vs ubuntu, but since Varnish restart fixes things I have to assume there is a cache that for some reason my local machine isn't getting?

Anyway, I enabled the cache in "Better WP Varnish" - "http://plane.ardupilot.com/wp-admin/options-general.php?page=better-wp-varnish". This makes no difference using above process - and if I tell it to clear the whole varnish cache it reports an error. 

Then I tried to enable total cache (http://ardupilot.com/wp-admin/network/admin.php?page=w3tc_general&w3tc_note=flush_varnish). When I try flush the whole cache problem persists, though the server says it has been successful.

Where should we look for the log of calls to clear the cache?

hamishwillee, 2015-02-05:
Final fix by Tridge - involved update to default.vcl and making sure purge requests went to varnish rather than localhost.

closed 2015-02-05 by hamishwillee