ArduPilot/ardupilot_wiki #39: ANGLE_MAX not displayed in auto-generated Copter parameters

hamishwillee, 2015-02-09:


hamishwillee, 2015-02-09:
This script is run to generate advanced parameters: https://github.com/diydrones/ardupilot/blob/master/Tools/scripts/build_parameters.sh

I ran the [param_parse.py](https://github.com/diydrones/ardupilot/blob/master/Tools/autotest/param_metadata/param_parse.py) script with verbose settings to get the list of parsed output and ANGLE_MAX was not in it. I checked the source PDE file (./ArduCopter/Parameters.pde) and it was present. The only difference I can see is that the variable is described as N ASCALAR while other variables are GSCALAR: 
ASCALAR(angle_max, "ANGLE_MAX",                 DEFAULT_ANGLE_MAX),

Waiting for confirmation from Craig as to whether this is intentional.

hamishwillee, 2015-08-12:
@rmackay9 Should the ANGLE_MAX parameter be documented in the Copter parameters?

rmackay9, 2015-08-12:
@hamishwillee, yes it should definitely be documented.  It's a pretty important parameter really.

So the ASCALAR is used to say that it's a member of the AP_Vehicle class.  My guess is you're right that the parser is not looking for ASCALAR and I'll bet we will find that the plane's ASCALAR parameters are also not being picked up (like AUTOTUNE_LEVEL).

I'm not much of an expert on that script although i guess I could try and figure it out.

hamishwillee, 2015-08-12:
@rmackay9 Thanks very much. I'll try fix this.

hamishwillee, 2015-08-13:
@rmackay9 After much wailing and gnashing of teeth, I have worked out that the script is fine. The problem is that the [@Range parameter here](https://github.com/diydrones/ardupilot/blob/master/ArduCopter/Parameters.cpp#L454) is missing a semicolon.

Assigning back to you to fix!

rmackay9, 2015-08-13:
well, that must have been tough to find.  fixed in master!

closed 2015-08-13 by rmackay9

rmackay9, 2015-08-13:
I did a review to see if any other parameters were affected by the exact same problem and found one, PILOT_TKOFF_DZ so fixed that one too.

hamishwillee, 2015-08-13:
@rmackay9 Thanks for fixing - and for doing the right thing and checking the others! Yes, this was a pain to find, because I started on the false assumption that the data was fine and the script was wrong.