ArduPilot/ardupilot_wiki #45: Layout: Make sidebar appear on left for 768pt(w) displays

billbonney, 2015-02-11:
Fix template to keep Side Navigation Bar on left-hand side of screen for tablets.

Also reduce the size of the main text font by a few points, or have options to decrease the size (see below)

[image]

hamishwillee, 2015-02-11:
Please check out the current page where I've removed the boilerplate heading: http://copter.ardupilot.com/wiki/flying-arducopter/

[image]

Now, given we're looking at how it "should be" do you still have a problem with the size of the main font?

Regards
Hamish

billbonney, 2015-02-11:
I still think the font can be smaller, If you look at the ardupilot.com on the screen grab, that is clear as day on my iPad. so somewhere between that size and what we have would work.

hamishwillee, 2015-02-12:
Current height is 40px - increased this from 33px (default for theme) because I thought it looked better. What do you think of 33px
http://copter.ardupilot.tridgell.net/wiki/common-install-mission-planner/
http://copter.ardupilot.tridgell.net/wiki/flying-arducopter/

How about on desktop? I could make this size conditional on the screen width too, though I'd prefer a height that works for all.

If you're happy with 33px I'll query the wiki editor group for an opinion then force it live.

billbonney, 2015-02-12:
The 33px high header is good. Though I was referring to the size of the string copter.ardupilot.com in the safari title area as a size font that is readable, yet much smaller than the current text on the page

hamishwillee, 2015-02-12:
I understand now. 

This is something we should discuss more broadly because small fonts are harder for (some) people to read but allow more text to be displayed.

That said, have a look at the .tridgell site now. I have only changed the title and basic font size (so lots of it is still not scaled). It triggers on 783px. Is this "about right" on your device?

billbonney, 2015-02-12:
I would say a bit small now, we need a small,medium,large font button. Now only if I could see the sidebar. I enabled that on my site by using a 3 column layout, but only one sidebar used.

[image]

hamishwillee, 2015-02-12:
I'd rather not use a button - I don't believe it ever really provides a good user experience. Best we find an acceptable size for mobile. I've made just a bit bigger ....

hamishwillee, 2015-09-08:
This was "fixed" (some months ago) by changing the viewport such that the sidebar will display in landscape on high resolution screens.

closed 2015-09-08 by hamishwillee