ArduPilot/ardupilot_wiki #51: Update PPM Encoder doc (common)

hamishwillee, 2015-02-19:
Thesse both cover the PPM encoder. 
* http://plane.ardupilot.com/wiki/common-ppm-encoder/
* http://plane.ardupilot.com/wiki/encoder-3/

Randy suggests:   

* Merge them with the common-ppm-encoder info being at the top and the encoder-3 stuff being below.  The top part of encoder-3 can be zapped, so just merge across “Setup Manual” and everything below.

* It should probably be added to the Optional Hardware section and there should probably be a link to this page from the “Connecting the radio receiver” pages for APM2 and Pixhawk.  http://copter.ardupilot.com/wiki/connecting-your-rc-input-and-motors/

* Randy says "I really don’t recommend people use the PPM encoder because it apparently introduces lag (at least 5 people have told me so so even though I haven’t seen it I believe them).". Perhaps add as warning

gmorph, 2015-02-19:
Randy's suggestions sound good to me.

Thanks

Grant

hamishwillee, 2015-03-05:
How do http://planner.ardupilot.com/wiki/common-ppm-encoder-apm2x-atmega32u2/ and http://planner.ardupilot.com/wiki/common-ppm-encoder-8-channel-standalone-encoder/ fit in then?

gmorph, 2015-03-05:
The first web page is talking about the PPM Encoder that is embedded in the
APM2.5/6 boards.  The second one is talking about the standalone PPM
Encoder board you can buy from the 3DR store.  They are sufficiently
different that we do need both web pages although I would suggest both are
hardly ever re-programmed.

Thanks

Grant

hamishwillee, 2015-03-05:
Thanks -  so I will make them sub pages of the appropriate hardware pages.

I've deleted http://plane.ardupilot.com/wiki/encoder-3/

I've updated http://plane.ardupilot.com/wiki/common-ppm-encoder/ - with the material from the above doc.
- One question. It says 
"The official source code is in the Git repository: <a href="http://code.google.com/p/ardupilot-mega/source/browse/#git/Tools/ArduPPM/ATMega328p.html">http://code.google.com/p/ardupilot-mega/source/browse/#git/Tools/ArduPPM/ATMega328p</a>. The standalone PPM Encoder needs the ATMega328p version."
Where is the source now - not on google as shown I'm fairly sure.

gmorph, 2015-04-05:
Source is here:

https://github.com/diydrones/ardupilot/tree/master/Tools/ArduPPM

Thanks, Grant.

hamishwillee, 2015-04-07:
Thank you. I have updated the link, and I have closed this as completed.

closed 2015-04-07 by hamishwillee