ArduPilot/ardupilot_wiki #52: Add grouping in Dev wiki for companion-computer topics

hamishwillee, 2015-02-19:
These are all related to companion computers - a growing field

http://dev.ardupilot.com/wiki/raspberry-pi-via-mavlink/
http://dev.ardupilot.com/wiki/odroid-via-mavlink/
http://dev.ardupilot.com/wiki/edison-for-drones/

It may be good to create a main entry on the TOC about using companion computers, and then link these from there. 

Alternatively, if we move to a categorisation/tagged approach, a category.

gmorph, 2015-02-24:
Yep a TOC landing page about using companion computers sounds great.

Thanks

Grant

hamishwillee, 2015-02-24:
How about this one: http://dev.ardupilot.com/wiki/companion-computers/

Anything else we need to say? Is this the right place on the menu?

gmorph, 2015-02-24:
I updated it - hope that was ok.

Yep, I reckon that's the right place in the menu's.

Thanks, Grant.

hamishwillee, 2015-02-25:
Hi Grant,

Thanks for the update - making it clear what is returned to the computer and what it can do with the information is a very important point. I played with the English a bit and removed the bit where I specified "pixhawk" (since the concept of a companion computer could theoretically be used with any controller, even though we only cover one).

I'm closing this as "done". Thanks for your help.

Regards
H

closed 2015-02-25 by hamishwillee

hamishwillee, 2015-02-28:
Closed then. Note, I also added BeagleBone to the list of companion computers.