ArduPilot/ardupilot_wiki #61: Fix broken links and images from removal of google code wiki

hamishwillee, 2015-03-03:
All broken links and images are visible in the Broken Links list in each wiki's dashboard

The images are available on my drop box. https://www.dropbox.com/s/7dy3b1fm0t9pkob/ArduPilotGoogleCode.zip?dl=0

I will look at this as it becomes a priority. If someone reading this chooses to help then THANKS! Please let me know here For My Information only.

hamishwillee, 2015-03-04:
Fixed these links in Mission Planner.

hamishwillee, 2015-03-05:
Fixed these links in Plane

zlite, 2015-03-05:
Thanks, Hamish! Just a few more to go...

hamishwillee, 2015-03-05:
You're most welcome. Though I can't help thinking this is slightly procrastination as some of my next tasks are hard to resolve in a completely "satisfying" manner.

Rover done, APM Planner 2 done.

hamishwillee, 2015-03-05:
I think that all the broken links are fixed. For images I have generally uploaded the replacement. For links I have found the wiki equivalent where it exists. 

In some cases there is no equivalent (ie a lot of APM1 stuff has been cleaned up) . In this case the link has been removed where reasonable. In some cases the whole linking "section" has been removed if that is required for the document to make sense.

The broken link checkers aren't giving me any more hits, but it is possible there are still some cases that have been missed. Feel free to notify me if more are discovered.

closed 2015-03-05 by hamishwillee