ArduPilot/ardupilot_wiki #66: APM no longer supports latest ArduCopter (only) firmware - add note.

hamishwillee, 2015-03-13:
The latest Copter firmware doesn’t fit on APM1 and APM2 boards (plane, rover and antenna tracker still fit).  In line with this, we’ve stopped building the copter firmware on the build server (aka firmware.diydrones.com) and if you try to uploading latest using the mission planner’s secret Ctrl-Q feature you’ll see a fail message

Add note to above effect and link to the last known good firmware and instructions on how to use it on APM to relevant docs. 

Question for Randy
1. What is the last firmware that can be used? http://firmware.diydrones.com/Copter/
2. Some of the places we might update with this information listed below. Any others you know of?
http://dev.ardupilot.com/wiki/building-the-code/building-ardupilot-with-arduino-windows/
http://dev.ardupilot.com/wiki/building-the-code/building-the-code-on-mac/
http://copter.ardupilot.com/wiki/initial-setup/common-load-firmware-apm/
http://copter.ardupilot.com/wiki/common-autopilots/common-apm25-and-26-overview/
http://copter.ardupilot.com/wiki/introduction/choosing-a-flight-controller/
3. Does this affect Traditional Heli? http://copter.ardupilot.com/wiki/traditional-helicopters/loading-the-code/

CraigElder, 2015-03-16:
1.  The last version is 3.2.1.  There might be a 3.2.2
2.  I think that probably covers all the places
3.  Yes 3.2.1 is the last version

hamishwillee, 2015-03-17:
Hi Craig

Where is the proper place to download 3.2.1? I can find it in "stable" (see [here](http://firmware.diydrones.com/Copter/stable/apm2-quad/git-version.txt) ) but presumably this will update to another version over time. 
If I look in older folders I can't tell which to use - for example last year [this one is labelled as "3.3dev"](http://firmware.diydrones.com/Copter/2014-11/2014-11-08-01:11/apm2-quad/git-version.txt).

The proposed message will look like the following if you are logged in/not logged in (respectively).
[image]

ANyway, the warning is what the bulk of users would see. Are you OK with that text if I roll it out more widely?

Regards
Hamish

CraigElder, 2015-03-17:
The firmwares are available here
http://firmware.diydrones.com/Copter/stable/

Perhaps what we can do are copy the APM1 and APM2 firmwares to the
downloads section of the wiki and provide a link to them there

zlite, 2015-03-17:
Agreed. A static link to legacy hardware support would be perfect.

-c

hamishwillee, 2015-03-18:
Good idea. I have created separate downloads for [APM2.x](http://ardupilot.com/downloads/?did=121) and [APM1.0](http://ardupilot.com/downloads/?did=120) - zip was too big otherwise. 

I have updated the following docs with the warning (for common docs this only appears in copter):
* http://copter.ardupilot.com/wiki/other-project-and-common-topics/common-autopilots/common-apm25-and-26-overview/
* http://dev.ardupilot.com/wiki/building-the-code/building-ardupilot-with-arduino-windows/
* http://dev.ardupilot.com/wiki/building-the-code/building-the-code-on-mac/
* http://copter.ardupilot.com/wiki/introduction/choosing-a-flight-controller/

The following two documents suggest using Mission Planner, which "automatically selects the correct latest firmware for your board". What will happen now - will the mission planner still supply the latest version that fits, or will no option be provided for copter when connected to an APM? Or will installation just fail (or something else)?

* http://copter.ardupilot.com/wiki/traditional-helicopters/loading-the-code/
* http://copter.ardupilot.com/wiki/initial-setup/common-load-firmware-apm/

rmackay9, 2015-03-18:
Thanks Hamish.
Yes, I think the mission planner will continue to provide the last version that fits on the board.  So for APM1/2 if they download with the mission planner they will get AC3.2.1.
It's probably totally clear but just in case, the current official build (AC3.2.1) still fits.  It's the next official version (AC3.3) that won't fit.

hamishwillee, 2015-03-18:
Hi Randy, 

Thanks for the clarification. I have changed "Current builds of the ArduCopter firmware no longer fit on APM boards. " TO  "ArduCopter 3.3 firmware (and later) no longer fits on APM boards." everywhere.

Also updated http://copter.ardupilot.com/wiki/initial-setup/common-load-firmware-apm/ message, but did not alter the [traditional helicopter page](http://copter.ardupilot.com/wiki/traditional-helicopters/loading-the-code/) - as that requires a lot more thorough review.

I believe this is closed from a wiki point of view - there may still be some "official announcement" to happen.

closed 2015-03-18 by hamishwillee

rmackay9, 2015-03-18:
@hamishwillee, great, thanks for that!

GippslandHobbyCave, 2016-05-26:
I am trying to set up ESC's through Mission Planner on an AdruPilot 2.8 - It is wanting V3.3.x of the firmware for a HexaCopter. I cannot use 3.2.1 as mission planner will not allow it - Any Ideas ?

rmackay9, 2016-05-26:
mission planner still allows downloading the older firmware.  It's possible that it doesn't support actually using the firmware although I'd be a bit surprised.  ESC calibration doesn't require the mission planner anyway though, the instructions are on the wiki and it's something like:
1. put TX throttle to full
2. plug in battery to flight controller and ESCs
3. unplug battery
4. plug in battery again
5. ESCs should beep, then reduce throttle to minimum and they should beep again

misiek08, 2016-05-30:
Is it possible for MP to don't show newer firmware than actually supported by hardware? I connected APM 2.6 and MP shows possibility to download AC 3.3.3. IIUC it won't work on this board.

rmackay9, 2016-05-30:
A reasonable request I'd say!  Maybe put it in the [mission planner issues list](https://github.com/ArduPilot/MissionPlanner/issues)?

sieder, 2016-09-19:
I am using APM 2.8, which ArduPlane framework is compatible with it?

magicrub, 2016-09-19:
Framework? This branch (last committed over a year ago) is the source code
that compiles for the old 8bit systems.

magicrub, 2016-09-19:
https://github.com/ArduPilot/ardupilot/tree/ArduPlane-release-AVR

gluca, 2017-01-08:
Dropping support for 8bit APM boards is only due to size of the build? Or there are some other issues related to the 8bit architecure and/or performances?
In other words is it possible to backport some of the 3.3 and 3.4 firmware features to the 3.2 branch?

gmorph, 2017-01-08:
For Copter and Plane its due to the size of the build.  We are down to bytes literally.
For Rover and AntennaTracker some features can be back ported and there are plans to do so.
EKF will never run on 8bit hardware - way too much math.  
Thanks, Grant.

magicrub, 2017-01-08:
The Arduino did not have enough program memory, RAM, and cpu speed. Basically, everything that makes Arduino an Arduino. The code branch is still there for your enjoyment but the developers have since moved on long ago.