ArduPilot/ardupilot_wiki #72: Add a Copy-To-Clipboard Feature for code setup

billbonney, 2015-03-22:
In regards to conversation about commands in code boxes on WP.
[2015-03-22, 07:50:29] Bill Bonney: The problem with splitting the boxes is that you need then to do that for all the instructions above that are multiline for consistency. My suggestion would be to put '$ ' prompt indicator at the beginning of each line OR write the lines as make configure && sim_vehicle.sh -w which executes the commands after each other
[2015-03-22, 07:52:06] Bill Bonney: (There also the concern how that impacts 'cut and paste' of code in the boxes. Maybe that's an idea, actually having the cut and paste box like used on github for example.
[2015-03-22, 08:56:59] Bill Bonney: here's a screenshot of the copy-to-clipboard functionality I was talking about

[image]

[2015-03-22, 08:59:44] Bill Bonney: and there's a plugin for that https://wordpress.org/plugins/tags/copy-to-clipboard :) (and probably a million others)
[2015-03-22, 09:00:53] Bill Bonney: (PS: that above one doesn't have the dependency on Adobe Flash like the github one)

hamishwillee, 2015-03-23:
Not as easy as you might think - or at least some decisions to be made.

* Most of these sorts of plugins require explicit markup, which creates a dependency on that plugin. 
* Do we want this everywhere - ie should we apply this to all code blocks, or is this to be applied "judiciously"

One alternative is to use

brunoolivieri, 2019-07-12:
Would this issue still make sense to stay open?  I believe that could be closed.

Hwurzburg, 2019-11-02:
This has been inactive for a long time...if anyone wishes to actively pursue this, reopen

closed 2019-11-02 by Hwurzburg