ArduPilot/ardupilot_wiki #73: Camera Control Board Doc is confusing

hamishwillee, 2015-03-23:
The overview does not sufficiently outline why this product has a point and what it offers. At the moment it talks a fair bit about CHDK, has a picture for contact closure (but doesn't explain what it is), talks about a picture protocol, but not what that means in terms of this product.

The overview needs to say more about the features and what they mean 
- what cameras/connections/protocols does it allow us to talk to
- what can't we do that we might expect to be able to?
- What if our camera supports IR rather than cable
- What do all those other connectors do?

The "Mavlink details" section is "code without context". As such I don't really know what the author thought I would do with this information?

I am envisaging that the benefit of this board is that we can talk to it in Mavlink using a generic set of commands for doing camera control. We connect to the ArduPilot using the UART, and to the camera using the specific port. Our MAVLINK commands then get translated to the specific target type, with some known behaviour if a particular command is not supported. This would make sense to me because it is the value proposition for the board - making a common interface for photography and simple connection to multiple target camera types. (and ideally extensible for camera specific functionality). I could be wrong of course, but either way, we need to make it clear what is actually does/is good for.

rmackay9, 2015-03-23:
It's slightly questionable that a product that's not available is listed on the wiki.  I guess it's mostly a page to recruit developers.  I'd say zap the final MAVLink section because I don't think that particular development detail is really necessary.  If someone is going to try and get involved in the development they'll contact craig and then there will be a ton of additional info they will need (and hardware) so that little snippet is just a small part of what they will need.

hamishwillee, 2015-03-23:
Yes, a bit questionable but otherwise we won't take it forward

I agree re the MAVLink section, but I won't update until I understand what is needed for the whole doc. 

Thanks.

auturgy, 2017-07-26:
@rmackay9 @hamishwillee Can be closed: dead issue

closed 2017-10-15 by auturgy