ArduPilot/ardupilot_wiki #74: Create single waypoint commands doc

hamishwillee, 2015-03-24:
There is this [ArduCopter Mission Command List](http://copter.ardupilot.com/wiki/mission-planning-and-analysis/mission-command-list) rather nice list of commands for arducopter and a (partial) generic list in [Planning a Mission with Waypoints and Events](http://copter.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/#basic_waypoint_commands). Other lists are scattered through the docs. 

The task is to create a single "common-Mission Command List" which lists all commands for Rover, Plane, Copter. It will need to clearly identify which are applicable/can be used in each platform, and any differences in meaning for each platform - for example takeoff might mean something difficult in Rover.

It can include commands that are not supported, but should clearly list the "introduced in" version.

Actual organisation to be decided - could be flat list, or be organised by functionality -e.g. Gimbal/Camera. Might even be "Common Commands", Rover specific, Plane Specific, Copter specific.

This will replace the copter mission list and the list above and be linked to from everywhere (for example this [3D Mapping](http://copter.ardupilot.com/wiki/common-3d-mapping/) topic).

hamishwillee, 2015-03-24:
I've split the commands in *ArduCopter Mission Command List*  and *Planning a Mission with Waypoints and Events* below to identify which are "common" and which are just documented in one list or the other.

Questions:
- Are all commands for ArduPilot vehicles in the lists below. If not, which are missing?
- Which of these commands do/do-not apply to Rover/Plane. I'm guessing most will apply to plane, but some like Spline waypoint might not apply to Rover?
- Are the copter list items actually common to all?
- Should the items in the last list be common to all
- Are there commands which have a "different meaning" on Plane/Rover?
- Are there any commands coming that might be in mission planner but are not implemented yet - if so what.
- Are there any commands for which you would expect the settings and map appearance in Mission Planner to be very different - ie if I use the Copter Mission command list images as a Rover example, is this still OK.

Commands in both lists
==================
TAKEOFF
WAYPOINT
LOITER_TIME
LOITER_TURNS
LOITER_UNLIM
RETURN_TO_LAUNCH
CONDITION_DELAY
CONDITION_DISTANCE
CONDITION_YAW
DO_JUMP
DO_CHANGE_SPEED
DO_SET_RELAY
DO_REPEAT_RELAY
DO_SET_SERVO
DO_REPEAT_SERVO
DO_SET_ROI
LAND

ArduCopter Mission Command List (only)
================================
Do-Digicam-Control
Do-Mount-Control
Spline Waypoint
Do-Set-Home
Do-Set-Cam-Trigg-Dist

Planning a Mission with Waypoints and Events (only)
================================

CONDITION_CHANGE_ALT
DO_SET_MODE
DO_CONTROL_VIDEO

hamishwillee, 2015-03-26:
Following discussions with Grant we think the best way to handle this would be similar to the ArduPilot parameters - a source definition that can be exported for different platforms. Grant is taking ownership and will discuss this with Tridge. I am happy to do the "Work" once the format is defined (either or both of the coding to extract the information and the work to put the information into a definition file).

hamishwillee, 2015-05-04:
First draft of the "manual" version here: http://planner.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/common-mavlink-mission-command-messages-mav_cmd/ 

@gmorph we need to review this (see my email) and also find time to discuss with Tridge if this should be automated.

hamishwillee, 2015-08-12:
@gmorph How you going with your checks to http://plane.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/common-mavlink-mission-command-messages-mav_cmd ? (friendly reminder!)

hamishwillee, 2015-09-08:
@gmorph Reminder on the list of items!

gmorph, 2015-09-09:
Its my top priority - on page 4 of my todo list ;-)

brunoolivieri, 2019-07-12:
Would this issue still make sense to stay open?

rmackay9, 2019-07-12:
We should definitely add the equivalent of Copter's support command list for Rover so I've changed the title.

Hwurzburg, 2019-10-10:
@rmackay9 cant this be closed now?

rmackay9, 2019-10-11:
@Hwurzburg, no, I think we need a Rover specific page which mentions all the mission commands rover supports and their effect.  We've got a very general page now that isn't easy to understand I think.  http://ardupilot.org/rover/docs/common-mavlink-mission-command-messages-mav_cmd.html

Hwurzburg, 2019-10-11:
@rmackay9 we need to discuss what you would like to see changed...discuss next call