ArduPilot/ardupilot_wiki #77: Get APM logos for Antennatracker and Dev sites

hamishwillee, 2015-04-01:
Currently these do not have logos. Have requested from MaxL

gmorph, 2016-11-15:
When we say logos in the new wiki do we mean the landing page picture?  i.e. in Rover there is a picture of a Rover but in AntennaTracker or Dev there isn't a picture of anything?

hamishwillee, 2016-11-15:
This is no longer an issue. In the old site each area had a special image in the top left corner identifying the site area. Now we just have the text "copter", "plane" etc.

I think that icons are nice, and we could add one. However getting icons produced and agreed on is just one more thing we don't really need to worry about now ... after all, the discussion about what the name of the autopilot is comes up every few months still ....

closed 2016-11-15 by hamishwillee