ArduPilot/ardupilot_wiki #78: Create article showing how to set up camera triggering using Skysight MONO

hamishwillee, 2015-04-02:
New board, discussed here: http://diydrones.com/forum/topics/using-aux-pins-as-relays-for-chdk?commentId=705844%3AComment%3A1953391&xg_source=msg_com_forum

hamishwillee, 2015-04-02:
Have created article: [Camera Shutter Triggering for Sony MultiPort Connectors using SkySight MONO](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-cameras-and-gimbals/common-camera-shutter-triggering-for-sony-multiport-connectors-using-skysight-mono/). This is published but not yet linked into menu.

Seeking review and approval to use the images.

gmorph, 2015-04-07:
A quick skim - it looks fine..  I can't test it as I don't have a SkySight
Mono.

Thanks, Grant.

hamishwillee, 2015-04-07:
Thank you . I can't test it for same reason. However I'm going to mark this as closed on the basis of anecdotal evidence it all works (ie forum messages) and the fact that the detail comes from their user guide, which you would hope works. 

I didn't get explicit permission to use the images or any comment on this, but given they used some on the forum and chose not to comment, I think this counts as "public domain".

closed 2015-04-07 by hamishwillee