ArduPilot/ardupilot_wiki #79: Search only works on subsites, not top-level ardupilot.com

zlite, 2015-04-06:
Search returns no results at the top level of ardupilot.com, but is fine at the subsite level (plane.ardupilot.com, copter.ardupilot.com, etc).

Needless to say, search at the top level should search across all the subsites.

hamishwillee, 2015-04-07:
I agree this is desirable. I do not consider it essential because most of our users are interested in a particular vehicle type and are more than likely to stick within that domain. 

None of the top 3 or 4 WP search plugins (including  [the one we use](https://searchwp.com/features/) ) support this feature. I have requested information from our vendor on whether this is on the roadmap.

There are a couple of other plugins that support global search.

* [MultiSite Global Search](https://wordpress.org/plugins/multisite-global-search/) - not many downloads, does not state support for our current version of WP, not highly recommended, does not look very configurable.
* [WPMUDEV Global Site Search](https://premium.wpmudev.org/project/global-site-search/) - commercial plugin - $25USD per month. Looks like it might do what we want.
* [WordPress MU Sitewide Tags Pages](https://wordpress.org/plugins/wordpress-mu-sitewide-tags/) sets up a hidden site with all content, and then can do some redirecting for searches. Not supported in latest versions, and a bit of a "hack". 

Other options

* Integrate a [google custom search](https://cse.google.com.au/cse/publicurl?cx=018052806301214587834:eoano1vurnc) for the multisite (and decide whether we can put up with adds or not). Note that I decided not to use a system like this originally because I dislike the advertising and cost was quite high for removing them.
* Remove search altogether on top site
* Replace the drop down on home site with a selector that forces you to choose a target site for the search, and redirects.

Chris, any particular opinions on this? The best approach depends on how important this feature is, and how annoying things like advertising might be. IMO this is not a high priority and I'd lean towards the replacement option last on bullet list above.

zlite, 2015-04-17:
Hamish, thanks for the research on this. Maybe the best thing is to just not display the search bar on the top level site. The problem now is that we let people search at the top site but display no results. Not giving them the option to search until they get to the sub-sites would be better.

hamishwillee, 2015-04-17:
Hi Chris

Thanks for the direction. I agree this is probably the most pragmatic decision - sorry I couldn't find a low-cost solution that properly meets the requirement :-(

I'm splitting my time between DroneKit and wiki at the moment, but I hope to look at this next week.

Regards
Hamish

hamishwillee, 2015-07-27:
This is now done - if the site domain name is "ardupilot.com" the search toggle is not shown.

The [relevant code changes are here](https://github.com/hamishwillee/ardupilot_theme_2/commit/bf41e53ed46336d2743d9fe2c2afec8368a1a62a). Note that this won't work on the home site in the staging server as it's domain is ardupilot.tridgell.net.

Closing !

closed 2015-07-27 by hamishwillee