ArduPilot/ardupilot_wiki #80: Create a Wiki page for Complex Mission Planning

paradisephill, 2015-04-07:
Over the last six months I’ve been struggling with understanding how mission commands are processed by the flight controller while I’m trying to capture an aerial panorama. The current Wiki on mission planning only addresses the basics.  After a great deal of trial and error and after finding a post from Soren Kuula I think I finally have a fair understanding of how complex missions are handled in Arducopter and I think adding the material to the Wiki would be helpful to others trying to create a complex mission.
I've added this material to my blog post at: http://diydrones.com/profiles/blogs/using-arducopter-3-2-missions-for-aerial-panoramas

Feel free to use any or all of it for the Wiki. 

Flying Complex Missions with “Condition” and “Do” Commands

When creating a complex mission with “Condition” and “Do” commands it’s helpful to have a good understanding of how mission commands are processed by Arducopter. You can think of the command processor as having two threads that run simultaneously. One thread processes “nav” commands and the other processes “non-nav” commands. However, it’s not a simple as just processing each command in order.
Think of a mission as being made up of a series of blocks of mission commands. Each block consists of an initial “nav” command (waypoint, loiter, etc.) and the following “non-nav” commands (“do” and “condition” commands).  So in Figure 1 think of commands 1 through 4 as Block 1, commands 5 through 7 as Block 2, and command 8 as starting the next block. A block can have many “non-nav” commands or it can be just a “nav” command with no “non-nav” commands. 

https://www.dropbox.com/s/9jpu2eybjimkrqq/Figure_1.jpg?dl=0
 
Figure 1
The command processor processes each block in the following way: 
1.	First execute the “nav” command. 
2.	When the vehicle is within WP_RADIUS centimeters (default = 200) of the desired waypoint, start the “nav” command delay timer, if any.
3.	When the “nav” command delay timer reaches zero or if there is no delay
a.	Stop processing “non-nav” commands from the previous block, even if they are not finished
b.	Start processing the “non-nav” commands in this block in sequence AND, AT THE SAME TIME start processing the “nav” command in the next block.

“Non-nav” commands that are “Do” commands process in a few milliseconds but commands that are “Condition” commands are processed until the conditions are met. In Figure 1 above, command 3, the CONDITION_YAW command, will be processed until the vehicle rotates to within two degrees of the desired yaw angle, at which time it will complete and Command 4, the DO_DIGICAM command, will be processed.
Now the processing gets interesting, notice that Command 5, a WAYPOINT command, has a 7 second delay. Remember in 3b above that the processing of the “non-nav” commands and the “nav” command in the next block starts at the same time. In order to allow enough time for the “non-nav’ commands of the first block to complete, you need to delay the completion of the “nav” command in the second block. Add up the time it takes for each “non-nav” command, add a small margin, and add it as a delay on the following “nav” command. Here are the delays I calculated for each “non-nav” command in the first block:
2 - 0 seconds
3 - 3 seconds to rotate from some arbitrary heading to 90 degrees due east
4 - 3 seconds for the CHDK script to initialize and take two images
Plus 1 second for margin
So command number 5, WAYPOINT, has a 7 second delay. You will need to do this math for every block in the mission that has “non-nav” commands. Even if every “non-nav” command is a DO command you will still need to add at least a one second delay when you use a “nav” command that completes right away such as a WAYPOINT with zero Lat, Lon, and Alt.
Using DO_JUMP
The DO_JUMP command is helpful when you want to repeat a sequence of commands multiple times. However, there is one characteristic of the command that you need to take into account, you cannot jump to another command in the same block! All jumps must be to a different block. There is an additional restriction that you cannot have more than three DO_JUMP commands in a mission in Arducopter Version 3.2.x. This is on the list to be increased in Version 3.3.
Consider this example:

https://www.dropbox.com/s/rj3ajyjq5z9r8l1/Figure_2.JPG?dl=0
 
Figure 2

Commands 1 – 10 are the first part of a mission to capture a 360 degree panorama.  
WAYPOINT  Command 1 starts Block 1. Command 2 sets the pitch angle of the camera gimbal and Command 3 points the entire copter due East. Command 4 takes the first photo. 
WAYPOINT  Command 5 starts Block 2. Command 6 rotates the copter by 28 degrees relative clockwise and Command 7 takes another photo. Command 8 starts the third block and allows enough time for the yaw to complete and the CHDK script to run after the DO_DIGICAM_CONTROL command.
WAYPOINT  Command 8 starts Block 3. Command 9, the DO_JUMP command, sends processing back to Command 6 in Block 2. In addition to moving the “non-nav” processing pointer to Command 6 it moves the “nav” processing pointer back to Command 8 and restarts the 5 second timer.  It repeats this 11 times to get the remaining photos for the first row of the panorama.
WAYPOINT  Command 10 starts Block 4. This block will change the pitch angle of the camera gimbal and subsequent blocks capture the remaining photos.
Here’s what the command processing looks like in the log, notice the interleaving of the “nav” and “non-nav” commands:

https://www.dropbox.com/s/5att2obgnooq2ep/Figure_3.jpg?dl=0
 
Figure 3

Timing of the CONDITION_YAW and the DO_DIGICAM_CONTROL commands
With limited air time (about 8-9 minutes) and a desire to do a 61 photo panorama, the amount of time taken for each photo is critical. Even one extra second per photo means another minute in the air so I did a lot of testing to minimize the amount of time taken in the main loop between commands 6 and 9 (which is repeated multiple times in the remainder of the mission, not shown here). I’ve shown in Figure 3 how I arrived at my timings so that others, with different sized frames, can develop their own timings. The time required to run the mission is about 5 minutes and 45 seconds. Combined with a takeoff to mission start of 45 seconds and a mission end to land of 1 minute and 55 seconds I have just enough time to get back on the ground before the battery voltage failsafe kicks in!

https://www.dropbox.com/s/15rm3rqh6dmc43x/Figure_4.jpg?dl=0
 
Figure 4
Figure 4 is a graph of the desired yaw angle and the actual yaw angle. The CONDITION_YAW relative command is initiated at time T0 which adds 28 degrees to the current Desired Yaw angle. At time T1 the frame has rotated to within 2 degrees of the desired yaw angle and processing of the CONDITION_YAW command is completed. The DO_DIGICAM_CONTROL is then executed. The CAM_DURATION  time for the DO_DIGICAM_CONTROL is .3 seconds. On my Canon camera, CHDK does not start until the shutter voltage is “released” so time T3 is when CHDK actually starts. Through separate testing I determined that it takes CHDK about 1 second to take the first photo and another second to take the second photo from that position (the second photo is insurance, I need the second one to substitute for the primary photo for one or two of the images in each mission due to blurring). So the total time required for the photos is almost 2.5 seconds. 
There are two big variables, the time for the frame to rotate to within 2 degrees of the desired Yaw Angle and the amount of time for the frame to settle back down after any overshoot. PID tuning and wind play a big factor in both. I’m using the default PIDS, the Autotune PIDs were two jerky for my purposes. As a result of all the testing I ended up using 3 seconds delay after the DO_DIGICAM_CONTROL command by either building it into the WAYPOINT delay as in this example or adding in a separate CONDITION_DELAY after each photo.

Yaw Stability

Wind is a problem for me because the gimbal has to be mounted forward of the centerline to balance the batteries. This creates an asymmetrical cross section that acts as a rudder in moderate (10 MPH) winds and tries to rotate the frame off the desired yaw setting. This can invoke a “feature” of Arducopter that changes the desired yaw setting if it can’t be attained within a few seconds. 

https://www.dropbox.com/s/861m9f5iw81pmtq/Figure_5.jpg?dl=0
 
Figure 5
[... cut ...]  yaw until it can. If you use a log series of CONDITION_YAW relative commands or you spend more than 10-20 seconds in one yaw position this can cause subsequent gaps in your panorama when you thought you had taken overlapping photos. I’ve minimized this in my mission plan by doing the panorama row by row instead of column by column, not staying on one yaw heading more than three seconds, and resetting the yaw of the copter back to an absolute yaw position at the beginning of every row of photos.

hamishwillee, 2015-04-07:
Thank you very much! I've assigned to me, but I won't have time for this in the short term, so if you would like to create as a wiki article yourself and have me review (or if anyone else would like to take this) then let me know.

paradisephill, 2015-04-07:
Thanks, Hamish.
I'm getting packed up to go on a motorhome trip to Arizona on Friday so I won't have time to do any more on it until early May. If you still need a Wiki author at that time I can work on it then if you can point to the style manual or equivalent..

hamishwillee, 2015-04-08:
The [style guide for the wikis is here](http://planner.ardupilot.com/wiki/common-appendix/common-editor-information-resource/#style_guidelines). I've tried to keep it simple, because I don't want people to be put off contributing by too many rules. I'll subedit anyway and ask questions about things that confuse me. You need to be signed up as a wiki editor - I can set you up when needed.

Have a great trip to Arizona!

brunoolivieri, 2019-07-12:
Would this issue still make sense to stay open?  As the wiki is much more complete nowadays, I believe that could be closed.

paradisephill, 2019-07-12:
Please close it. With all the anti-drone laws in California I quit the hobby several years ago so this probably doesn’t apply any more.

closed 2019-07-12 by paradisephill