ArduPilot/ardupilot_wiki #81: Sort out the Maxbotix sonar wiki pages.

gmorph, 2015-04-09:
Hi Grant

I've just finished copying the common optional hardware across to Rover.

Then I discovered this article, which is at the end of the rangefinders; Sonar Sensors, and which provides Rover specific setup of http://rover.ardupilot.com/wiki/common-optional-hardware/common-rangefinder-landingpage/common-rangefinder-maxbotixi2c/ for APM2

I was wondering if there is any way to combine them or make the common doc more complete so that the Rover specific document can be smaller.
The Rover Sensor doc covers APM2, rather than pixhawk and additionally talks about setting up a second sensor.

My thoughts are
- common-Page - add connection information for APM2 from the rover doc
- common-Page - add "connect a second sensor" information? Is this desirable on plane/copter?
- common-Page - link to the rover specific page.
- Rover doc - make a sub page of the common docs
- Rover doc - rename to make it clear that this is Rover specific and covers APM2 only

Thoughts? Any other ideas how common vs specific case here should be done? A KISS solution might simply be to rename the Rover doc appropriately.

Cheerio

gmorph, 2016-11-15:
The Maxbotic I2C and Analog rangefinder pages are kept separate as the physical installation and setup of these devices is a very separate process.  The testing of the sensor is the same i.e. the monitoring of the status of sonarrange and sonarvoltage but everything else needs to remain separate. 
I'm not sure about the statement "The Rover Sensor doc covers APM2" as the Analog page covers APM2 and Pixhawk and the I2C page covers pixhawk and these pages are the same for all vehicle types.
Closing for now as I think the doco for these is good but if we want to revisit happy to reopen and discuss further.
Thanks, Grant.

closed 2016-11-15 by gmorph

hamishwillee, 2016-11-16:
Your call :-) FWIW, I agree that if there is separate setup then pages should be separate. Whether or not you were to split out the common testing would depend on how much of this there was (if a significant amount of duplication, then yes, that would be split into a separate page.)