ArduPilot/ardupilot_wiki #82: Fix inconsistent Ardu/APM branding in wiki text

hamishwillee, 2015-04-09:
The following changes will be made to reflect our autopilot branding:

* Change all instances of APM:Pilot to ArduPilot
* Change all instances of APM:Copter and ArduCopter to "Copter"
* Change all instances of APM:Plane and ArduPlane to "Plane"
* Change all instances of APM:Rover and ArduRover to "Rover"
* AntennaTracker referred to as "Antenna Tracker" (TBD)

The official name of copter/rover/plane is still APM:Copter (etc) but this will only be used in the rare cases it is needed (e.g. referring to the platforms outside of the site). 

Note:
* Changing instances where ArduPilot is referred to as APM cannot be automated, and those changes will have to be made on an ad-hoc basis.

rmackay9, 2015-04-09:
Maybe just shorten AntennaTracker to just Tracker.

CraigElder, 2015-04-10:
Also please ensure that the capitalization of ArduPilot is consistent throughout the wiki.

hamishwillee, 2015-04-10:
Will do my best!

hamishwillee, 2015-04-10:
What are official names for "Mission Planner" and "APM Mission Planner 2"  going to be?

gmorph, 2015-04-10:
We haven't discussed changing the names for those.  The respective
maintainers would need to be involved which hasn't been arranged so they
will remain un-touched for the moment.

Thanks, Grant.

hamishwillee, 2015-04-10:
Possible problem pages

* http://rover.ardupilot.com/wiki/apmrover2-parameters/ - name may be hard linked in auto scripts

hamishwillee, 2015-04-10:
This is done. I will reply with respect to "limitations" on the ardu wiki editors group.

closed 2015-04-10 by hamishwillee