ArduPilot/ardupilot_wiki #91: Separate wiki by platform types , i.e. APM, PixHawk, Linux

lvale, 2015-05-19:
Currently the wiki has all platforms mixed, either by having separate articles for APM and for PixHawk - example are:

http://copter.ardupilot.com/wiki/common-optional-hardware/common-telemetry-landingpage/connecting-your-rc-input-and-motors/

http://copter.ardupilot.com/wiki/common-pixhawk-wiring-and-quick-start/

Or articles that don't refer to which platform the feature is applied like:

http://copter.ardupilot.com/wiki/configuration/ekf-inav-failsafe/

Or pages that mix instructions for APM and PixHawk like

http://copter.ardupilot.com/wiki/common-optional-hardware/common-rangefinder-landingpage/common-rangefinder-maxbotix-analog/

For the sake of clarity, I believe these should have a top level filter/division to enable users to see only the applicable articles to each platform, APM/PixHawk/Linux/...etc...while maintaining the non platform specific articles on each tree repeated, like what is done now for plane/copter/rover, etc.

So the top level would be Copter/PixHawk, or Copter/APM, or Plane/PixHawk Plane/APM.

When new platforms appear the existing structure would be able to cope with them, and create a specific tree to that platform.

lvale, 2015-05-19:
Hi @hamishwillee 

Regarding the comments of your reply from the other repo (sorry for my mistake:) ):

For a novice user, I believe that a complete separation by platform is easier, even if it implies a duplication of articles when these are platform agnostic, and specially on those sections (like you refer) where there are only minor differences. If there are differences (minor or major) that should be enough to have different articles by FC. And on the optional Hardware is specially relevant to have them separate.

I think of the wiki, like a self contained book. You read the APM volume, the Linux volume or the PixHawk volume, and these don't refer each other. They are "self contained". In the beginning it would require a larger amount of work to separate some of the articles, but at this point, most of the common stuff will be a copy/paste/minor edit job.

The Optional Hardware case you mention is one that I would have completely separate. People try to find  how to connect a PixHawk to a Sonar, or whatever gizmo, and if we present them with clear and "uncluttered" instructions to do it, they'll have less chance of mistake and clearly improves the readability.

Your comments (from the other repo) quoted below for simplicity and context. I'll edit them from here after you copy them here :)

----------------------------------------------------------------------------------------------------------------------------
Hi @lvale

Thanks for your post - much appreciated. Just FYI the right place for wiki issues is here - this repo just holds source to make it easier for Chinese sites to translate our content.

Yes, the organisation is a bit haphazard. I am in broad agreement that the sections under Assembly Instructions should be better organised by platform - and this would also help us recognise where we are missing material on different platforms. This is something I hope to get to as I continue moving to just one TOC (a tedious process that I am doing in parallel with work on DroneKit).

This will not be something as rigid as you are describing because we have a lot of topics where the flight controller is largely irrelevant or has lot of common behaviour with only small differences for flight controller. I really don't think it is smart to duplicate the content, and if you split the content and link to the controller-specific material then the instructions can become harder to follow.

What I would like to do is:

add metadata (tags?) to mark up what documents use what controllers
ensure that if a topic is controller specific that is mentioned in the title
Have commonly named ans structured topics for covering common material - so for example we might have an APM2.x QuickStart and that would cover the same topics and main areas as the PIxhawk Quick Start.
I also hope to keep the Optional Hardware to be platform agnostic and have separate sections in the document about connecting to different hardware.

Hope that makes sense?

hamishwillee, 2015-05-19:
----------------------------------------------------------------------------------------------------------------------------
Hi @lvale

Thanks for your post - much appreciated. Just FYI the right place for wiki issues is here - this repo just holds source to make it easier for Chinese sites to translate our content.

Yes, the organisation is a bit haphazard. I am in broad agreement that the sections under Assembly Instructions should be better organised by platform - and this would also help us recognise where we are missing material on different platforms. This is something I hope to get to as I continue moving to just one TOC (a tedious process that I am doing in parallel with work on DroneKit).

This will not be something as rigid as you are describing because we have a lot of topics where the flight controller is largely irrelevant or has lot of common behaviour with only small differences for flight controller. I really don't think it is smart to duplicate the content, and if you split the content and link to the controller-specific material then the instructions can become harder to follow.

What I would like to do is:

add metadata (tags?) to mark up what documents use what controllers
ensure that if a topic is controller specific that is mentioned in the title
Have commonly named ans structured topics for covering common material - so for example we might have an APM2.x QuickStart and that would cover the same topics and main areas as the PIxhawk Quick Start.
I also hope to keep the Optional Hardware to be platform agnostic and have separate sections in the document about connecting to different hardware.

Hope that makes sense?

hamishwillee, 2015-05-19:
Hi @lvale 

I completely agree that a focused single docset on a topic is easier for users to read and navigate. However it is much harder to organise in the first place and maintain. There is also more work in maintaining the clunky wordpress menu system to maintain this stuff. In other words, yes, but not without significant cost.

I think the relatively minor separation I propose above would give 80% of the benefit with 20% of the short term work and ongoing maintenance. Of course when I actually start that job, I might discover that you're absolutely right, and a lot more needs to be done. 

In any case, to proceed this we need the broad agreement of @rmackay9 and the wiki editors group. 

If they say yes, are you volunteering to do some of the the migration work?

lvale, 2015-05-19:
@hamishwillee 
Damn.... I should learn to keep my mouth shut :)

I believe that a "smooth" approach to the issue will return to bite us harder in the future when new platforms appear. On the other hand a "radical cut" by platform will be painful in the beginning, but much easier to maintain going forward.

I don't know enough of the "infrastructure" of the wordpress/wiki system to know if a good taxonomy using metadata would be enough to get us the kind of "filtering" I propose. But first step would be the definition of that taxonomical structure.....(does it show I've worked with CMS systems before ? :) )

Yesterday at the dev call Randy was receptive...

I'll have to learn this wordpress/wiki stuff....Any pointers ?

hamishwillee, 2015-05-19:
Hi @lvale 

Hey, if you know what you're doing and you have the time to follow through end-to-end then keep the mouth open as much as you like. The problem is not with the idea, its the resource and the infrastructure. 

Wordpress is unlikely to provide the level of filtereing you'd want. Basically you can have flat lists of categories (hierarchical) or tags (flat). You can add other custom hierarchy which are basically the same thing, but for your own custom purpose. There isn't any way to dynamically rebuild a menu based on a particular category/tag.

I suspect that you're just going to have to work with the main instructions menu as I have done.

I have been thinking about metadata mostly for document maintenance purposes - ie to identify APM1.0 docs so that when a platform goes out of use we can tidy/remove those docs.

Defining what metadata we want is a good idea. I would follow KISS - pick the bits of dublin core or whatever that actually make sense. Some of the metadata is present, but not well handled in a wiki sense, since Wordpress is really not a wiki (despite the fact we call that) - specifically a real wiki has easy means for readers to work out who the contributors are.

Learning wordpress is not that hard as there are lots of docs out there. Perhaps if I give you access to the staging server so you can play with what we have now. Note, I might break this access later by accident :-0 When you've had a play and understand more how things work we can talk further?

As I said, I am working on an approach here. I think that we could move towards what you want in steps. Mine first :-)

Regards
H

hamishwillee, 2015-05-20:
Additional information. Topics can appear multiple places in the Table-of-contents. Typically we add once as a "wiki" entry and once as a "link" (though this has just been a convention, and there is no reason both can't be "wiki" entries). However obviously if you have a topic in 2 places you lose "context" given by the highlighting of the current topic in the sidebar - if you are in APM you might see the Pixhawk tree highlighted.

The Menu has historically been "flakey" - edits could break the wiki. I believe we have these fixed now :-0

lvale, 2015-05-20:
I believe this wordpress is not "plain" i.e. it has some plugins/extensions ?

hamishwillee, 2015-05-20:
Yes. Heaps. The wiki is a little unstable. The server experts are supposed to be looking at that. In the meantime I have got a restarter which will automatically bring the wiki up again in 1-4 minutes. Try again :-)

hamishwillee, 2015-05-20:
So a bit of a ramble about some options ....

The stated problems are that 

* Users are exposed to documents that might not be relevant. 
* It would be easier for users to just be able to find setup information relevant to their controller
* For future proofing, it is easier to add and organise material for new controllers if we already organise material by controller. 

I agree with all that, but we need to consider the practicality of different split options.

Consider for example [common-installing-3dr-ublox-gps-compass-module](http://copter.ardupilot.com/wiki/common-optional-hardware/common-positioning-landing-page/common-installing-3dr-ublox-gps-compass-module/). This appears both in optional hardware and in Assembly instructions. This has instructions for connecting to the different autopilots, but also has common material for mounting it, setting up mission planner and the LED indicators.

The suggestion is that  we need completely separate sections for the Pixhawk/APM/PX4 etc. There are some alternatives:

*  Duplicate this document three times (obviously removing the APM section in Pixhawk topic). I don't like that because then we have to maintain three copies of essentially the same material. We have also got three more topics to put in the menu, which is hard to maintain and notoriously flakey. Multiply that by all our topics which might need to be in a controller specific section and things are very messy.

* Link the same document for each platform. The doc will have redundant setup information for users of other controllers. This is probably the easiest approach, but we do lose the context in the sidebar as the topic will he highlighted in all the controller sections. We also have additional complexity in the menu. However this will mean that non-relevant stuff doesn't need to be maintained.

* Split this into four documents - the common stuff and three about "connecting" for the different controllers. The connecting topics would link to the common material. This has some duplication still and now the user who wants to know everything about setup of this component will have to go to another document to find out how to mount and do mission planner configuration. Also more complexity in menu.

* Split into multiple documents: about, connecting (per platform), mission planner configuration, mounting, etc. String the topics together by links. Have parent topic (about) with the other sub topics down in the optional hardware section. In the "top" setup we link to all these bits as needed. Probably even better, we have just the topic 

* Split into multiple documents as above and have them all in the one place down under the controller. So all controller specific stuff lives with the controller and links back. In the first time setup we'd link to the controller section. 

What we do now isn't perfect, but it is a reasonable compromise. IMO duplicating, multiple links and splitting all have their own problems. I don't really see adding new controllers as a big problem in this structure. I do like the idea of splitting docs much more finely, but I don't think our infrastructure is really up to it (the menus scary!).

What we clearly do need to do is be better at organising. Platform specific instructions should be grouped in the menu and if they are limited to a platform that should be clearly identified. 

If we are going to do anything re structure we need to do it thoroughly and globally, and have an extensible and scalable approach. We should prototype this "a bit" on staging server.

hamishwillee, 2015-05-20:
FYI, I just made one minor change in this area. The [Assembly Instructions](http://copter.ardupilot.com/wiki/initial-setup/assembly-instructions/) previously contained content on [Connecting the APM2](http://copter.ardupilot.com/wiki/initial-setup/assembly-instructions/connecting-the-apm2/). This didn't "work" because the top level grouping item was APM2 specific.

So I created the new document and turned the Assembly Instructions into a grouping page.  I also updated the menu so that all the topics for APM and Pixhawk are together in the menu.

This doesn't change the need to have the discussion above.

hamishwillee, 2015-08-12:
@lvale So we agreed that having separation into different autopilots is "a good thing" . Following discussion with Randy what we've done here is to make this separation at the "Assembly" level - you can see the grouping topic for [pixhawk here](http://copter.ardupilot.com/wiki/common-pixhawk-wiring-and-quick-start/).

I don't think it is worth pushing this division further up the wiki sidebar hierarchy because of the cost of duplicating content - am therefore closing.

Thanks very much.

closed 2015-08-12 by hamishwillee