ArduPilot/ardupilot_wiki #92: Investigate re-enabling wiki comments

hamishwillee, 2015-05-30:
Having comments against the wiki is the best place for discussions, and making sure that suggestions can be managed.

The comments are currently disabled (due to technical issues - possibly arising from introduction of two-phase authentication).

Action is to investigate possibility of re-enabling:
* Anyone should be able to comment, but not "anonymously" (not just diydrones/wiki editors)
* Ideally should accept DIY drones or wiki logins

hamishwillee, 2015-05-30:
I will need to investigate this. 

The requirement leads me to consider [Disqus](https://disqus.com/), a third party commenting solution that allows you to add comments to any website. This would completely separate the comments handling from our wiki infrastructure and authentication, bypassing the reason for removing comments in the first place.

The reasons against external comments systems tend to be around control over visibility, branding, ownership of the content and systems stability. IMO none of these are a concern here, and Disqus might be a good match. 

Disqus has excellent spam handing behaviour etc. 

Thoughts welcome - @rmackay9 @gmorph @tridge @zlite @tridge

rmackay9, 2015-05-30:
I think it would be good to try re-enabling he comments within the wiki and the Disqus sounds ok to me.  We use to have this and people used it quite a bit.  It's just a much more direct way to voice misunderstandings about the documentation.

hamishwillee, 2015-05-30:
I agree - I was quite surprised we didn't have the comments enabled when I first saw this. The only problem with Disqus is same as translation - URLs can't move. I may need to remove the url hierarchy I was playing with before we add this.

hamishwillee, 2015-06-01:
I had a further look at commenting using the native system. I don't see any way of making commenting both open and safe - the system can request an email address but not check it. Within about 20 seconds of enabling discussion on the staging wiki some spam appeared! We can of course make it open only to logged in users, but that is very restrictive.

@rmackay9 I have done minimum work to enable disqus on http://copter.ardupilot.tridgell.net/wiki/acro-mode/#disqus_thread

I think this is fairly effective. It would help me if you could create a couple of posts to the above link, the first should just be some words. Please add a few links in the second one.

What I have discovered if we use this
1. Will need to fix the commenting area on the theme - the comments currently too wide
2. The comment disabling system is still needed in order to disable comments for posts/pages. This keeps comments off our home pages.
3. The plugin respects the wiki commenting settings. This means a lot of pages will initially NOT have comments, because comments aren't enabled. This means two actions - making comments turned on by default, and adding comments for existing pages.
4. Pages can be moved and the comments move with them.
5. I will need volunteer moderators for each wiki. Initially these will be the same people as own the wiki. They will have to have disqus accounts.

Randy, any thoughts about the above?

rmackay9, 2015-06-01:
it looks really good to me.  I like the new features including:
  - nested replies
  - ability to vote items up/down
  - ability to post with google login (amongst others)
The only thing that seems missing is the ability to delete my own comments.

hamishwillee, 2015-06-02:
Thanks very much. I like nested replies. Not sure about value of voting, though I guess it can't hurt! 

I believe you can delete your own comments by hovering over the little icon as shown in image below. This may only apply to registered "Disqus" accounts - let me know if you get the option please?
[image]

hamishwillee, 2015-06-02:
OK, so I've played with this a bit and I'm fairly sure that I can make it work - enable comments globally across wiki items (both old articles and on by default for new articles going forward). I can make the layout OK on both desktop and mobile. A good example is here: http://copter.ardupilot.tridgell.net/wiki/acro-mode/

I'm going to have one more look at commenting. I found this plugin which may be worth looking at https://wordpress.org/plugins/inline-comments/screenshots/ given that we're a doc platform.

hamishwillee, 2015-06-02:
OK, inline comments didn't work well "out of the box". I think we go with Disqus. I'll request comment from the ardupilot editors group.

gmorph, 2016-11-15:
Is this discussion still relevant given we have a new Wiki system?

hamishwillee, 2016-11-15:
Essentially the question is more "should we support comments on wiki pages". Craig was opposed due to the cost of filtering out spam etc. I was supportive because I think discussion is the lifeblood of good docs. That said, I don't have time to monitor or maintain this currently, so perhaps we should close this until someone with time says that they are happy to monitor it.

gmorph, 2016-11-18:
I'm going to close this as I am hoping the new gitter room for Wiki discussions will fit the bill.  If not we can always come back and visit this issue again.

closed 2016-11-18 by gmorph

hamishwillee, 2016-11-18:
Agreed. Though should we add link to gitter channel on the docs themselves?