ArduPilot/ardupilot_wiki #93: Document AntennaTracker simulation in SITL

hamishwillee, 2015-06-04:
Document Antenna Tracker Simulation:

1. How to start a virtual antenna tracker
2. How to use  a real antenna tracker with a virtual aircraft (SITL)

Information below from @tridge in drones-discuss:

> Are there also any unit-tests cpp files for the AntennaTracker?

not unit tests as such, but you can start an antenna tracker in SITL
like this:

```
sim_vehicle.sh -T -j8 --console --map
```
the -T means to start a virtual antenna tracker as well as the virtual
aircraft. That should work on both windows and Linux SITL.

The antenna tracker simulator is [here](https://github.com/diydrones/ardupilot/blob/master/libraries/SITL/SIM_Tracker.cpp).

It can simulate either position servos or onoff servos. It can't simulate rate servos yet. Patches would be welcome for that.

If you are working on the AntennaTracker code I'd suggest you modify the simulator to match your real tracker, then use that for testing.

You can also use your real antennatracker with a virtual SITL aircraft. I found that very useful when initially developing the code. I setup SITL to fly around my hours (my virtual house!) then setup my real
antennatracker to point at the virtual aircraft. That allowed me to easily test the code and the mechanism.

Gabriele1995, 2015-12-07:
I have a real antenna tracker in my office. I would like to make it track virtual plane. How to make 3DR Radio to talk to Sitl?

hamishwillee, 2015-12-07:
@Gabriele1995 This area is for issues with the wiki, not tech support - suggest you ask this question on the APM forums.

rmackay9, 2015-12-07:
@Gabriele1995, i don't know how to do that exactly but it should be possible to make SITL forward on packets from the vehicle to a COM port on the machine which is connected to the tracker.  maybe check sim_vehicle.py ?

gmorph, 2016-11-15:
We don't normally document how to run SITL for a specific vehicle type.  However within the Dev wiki we have a whole SITL section.   I'll try to add a small amount of doco on how to do it in here:
http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html
Thanks, Grant.

hamishwillee, 2016-11-16:
We show how to run SITL for copter/plan, and rover is the same. The antenna tracker is different though - its not even clear what simulating an AT should "mean". So IMO this is worth documenting.  The description section above is supposed to provide enough information to work out how to answer the "what does it mean to simulate AT" and "how you do it"

brunoolivieri, 2019-07-12:
This Issue seems a discussion for the forums. Would this issue still make sense to stay open?

rmackay9, 2019-07-12:
I think it is documented somewhere on the wiki but I'm not sure.  I can't immediately find it: http://ardupilot.org/dev/docs/simulation-2.html.

We should leave this open until we confirm we have details on how to start tracker in SITL.