ArduPilot/ardupilot_wiki #94: Please add http://www.auav.co/product-p/auavx2_6.htm to the list of official supported boards listed on http://dev.ardupilot.com/

CraigElder, 2015-06-05:


rmackay9, 2015-06-06:
I have to ask though, where do we draw the line on listing all boards that run ardupilot?  We don't have the hobbyking boards or bangood boards for example.

Maybe we should only add boards that come from dronecode members or boards that we think add some new technical ability that enhances the project?

On my side, I think we should probaly add the NAVIO+ board as well.
http://www.emlid.com/shop/navio-plus/

For whatever board we add, I think we should keep the information to a minimum and just link to those retailer's wikis where they can describe how to connect all the various sensors to the board.

hamishwillee, 2015-06-06:
@rmackay9 Good question. It is (or will be) a DroneCode wiki, so I would expect that only boards from members would be added in any detail (by us). I certainly agree that we should encourage board owners to create their own documents and that we should link to that rather than duplicate. 

So what gets the "full treatment" on the wiki? I guess that would be the set of components that we consider either the be the "hardware reference" or very popular. 

What is standard information we would want in a table if we were to create a "big list" of these? What information would be enough to find the interesting links by searching the wiki?

CraigElder, 2015-06-06:
This is the Ardupilot wiki and there are more companies involved than just 3D Robotics now.  We draw the line with companies who are in compliance with the Open Source licences.  Hobby King and Banggood of course are violating the licences but I have been in contact with them about that.

@rmackay9 NAV IO is already listed as a supported board and I have invited them to join the Dronecode Foundation as a member.

Erle Robotics is already a member of the foundation and we should probably update the information for the PX4Fire / BBB cape.

ARSOV RC is also now in compliance with our licenses and their design files are available at 
http://www.auav.co/product-p/auavx2_6.htm 
They are also a new member of the foundation and they have been added to the list of recognized vendor certificates
https://github.com/diydrones/apm_planner/commit/0b59fee7b42390d75922481e91bd81a9828ae827
https://github.com/diydrones/MissionPlanner/commit/45f2ee4924a29032ffdd05ef9aa92b5f75acaca9

rmackay9, 2015-06-06:
@hamishwillee, for a big list.. I'd say we'd need:
1. where they can be purchased
2. operating system, cpu speed, RAM
3. link to vendor wiki describing setup
4. link to where design files can be found

@CraigElder, Ok, I'm convinced.  I like the key point being that the board design files are published.  That's certainly a clear line.

hamishwillee, 2015-06-07:
@CraigElder I added the board as you suggested [on dev wiki home page](http://dev.ardupilot.com/#supported_autopilot_boards). 

We should also add information about how to [build for it here](http://dev.ardupilot.com/wiki/learning-the-ardupilot-codebase/learning-ardupilot-introduction/#build_system) but there is no specific .mk file. Does the fact that it is "100% compatible" mean we can build using "make px4-v2 – the Pixhawk"?

There doesn't appear to be any vendor-wiki information on setup in this case?

Still thinking about a "big list".

hamishwillee, 2015-06-08:
@CraigElder @rmackay9 
The benefit of a list is that if we have standard slots for required information it is obvious when that information hasn't been provided. On the other hand, clearly no one has been "crying out for this".

I've "started" to think about how a "big list" might look (see prototyping http://dev.ardupilot.com/?post_type=wiki&p=5670&preview=true )

[image]

There is a lot of duplication in the specs information vs information in the product page and wiki. I am assuming therefore that having this here is a good thing for comparison purposes. Does this look like the right level of information (see link above). Is it worth expanding this or drop the idea?

I will need your help populating the information for the other boards.

CraigElder, 2015-06-08:
Hamish,
The board is electrically identical to a Pixhawk. You can load the regular
px4-v2 code into it directly from Mission Planner or APM planner 2.0
Building for it is identical to the Pixhawk.

hamishwillee, 2015-06-08:
@CraigElder - thanks - added a clarification on the [learning page here](http://dev.ardupilot.com/wp-admin/revision.php?revision=5187): 
> make px4-v2 – the Pixhawk **(and Arsov AUAV-X2)**

hamishwillee, 2015-06-09:
@CraigElder @rmackay9 
OK, I have filled the "big page" out a bit more. [Can you check it](http://dev.ardupilot.com/?post_type=wiki&p=5670&preview=true) and help me fill in missing bits? Is this useful?

rmackay9, 2015-06-09:
I think we should add an "OS" section.  So for AVR is would be "None".  Pixhawk would be NuttX,   Navio & Beaglebone would be Linux Debian (I think).

CPU for the APM2.x boards is the AtMega 2560.

hamishwillee, 2015-06-09:
Thanks Randy. OK, I've [created the page here](http://dev.ardupilot.com/wiki/supported-autopilot-controller-boards/) with the OS field. I'll plumb it into the wiki menu and ask for feedback from the wiki editors.

I'll close this issue as done though.

closed 2015-06-09 by hamishwillee