ArduPilot/ardupilot_wiki #95: Create doc listing all/common ArduPilot error messages

hamishwillee, 2015-06-10:
In particular, report any error/message that can show up on the Mission Planner HUD, for example if a failsafe is engaged.

This should appear in the menu around the first flight/getting started with flying information. Possibly part of Mission Planner docs - depends how generic messages are.

Some context in [this discussion](http://diydrones.com/forum/topics/copter-reliability-vs-functionality?commentId=705844%3AComment%3A2016929&xg_source=msg_com_forum)

brunoolivieri, 2019-07-12:
Would this issue still make sense to stay open?  As the wiki is much more complete nowadays, I believe that could be closed.

rmackay9, 2019-07-12:
we have a list (perhaps incomplete) of [pre-arm check warnings](http://ardupilot.org/copter/docs/prearm_safety_check.html).  We don't have a page with inflight errors though so this should probably stay open.

Hwurzburg, 2019-11-02:
so here is a list of severe messages (there may be ones that need work that are coded as INFO or other?), I removed Pre-Arms already covered, and obvious ones, like during calibration setup, or a result of a pilot action (like saving trim)...I can add the pre-arm ones, but do not know under which pre-arm check they fall, so need input on that....also, which of these warrant some kind of explanation or trouble-shooting hint...the list fairly long, you might want to copy and past into an editor to view it easier...

APMrover2/AP_MotorsUGV.cpp:            				 "PreArm: regular AND skid steering configured");
APMrover2/AP_MotorsUGV.cpp:            				 "PreArm: check skid steering config");
APMrover2/AP_MotorsUGV.cpp:            				 "PreArm: check steering and throttle config");
APMrover2/AP_MotorsUGV.cpp:                 			 "PreArm: servo function %u unassigned", function);
APMrover2/ekf_check.cpp:                    			 "EKF variance");
ArduCopter/AP_Arming.cpp:        				 "Arm: Motor Emergency Stopped");
ArduCopter/ekf_check.cpp:                    			 "EKF variance");
ArduCopter/ekf_check.cpp:               			 "Vibration compensation OFF");
ArduCopter/ekf_check.cpp:            				 "Vibration compensation ON");
ArduCopter/events.cpp:    					 "Failsafe: Terrain data missing");
ArduCopter/events.cpp:             				 "GPS Glitch");
ArduCopter/events.cpp:           				 "GPS Glitch cleared");
ArduCopter/mode_auto.cpp:           			   	 "Auto: Missing Takeoff Cmd");
ArduCopter/mode_rtl.cpp:        				 "Restarting RTL - Terrain data missing");
ArduPlane/is_flying.cpp:                			 "Hard landing detected. No action taken");
ArduPlane/is_flying.cpp:                			 "Hard landing detected");
ArduPlane/parachute.cpp:        				 "Parachute: Released again");
ArduPlane/parachute.cpp:        				 "Parachute: Released");
ArduPlane/quadplane.cpp:        				 "Transition failed, exceeded time limit");
ArduPlane/quadplane.cpp:         			         "QuadPlane mode refused");
ArduPlane/quadplane.cpp:        				 "Failed to complete takeoff within time limit");
ArduPlane/quadplane.cpp:        				 "Failed to complete takeoff, excessive wind");
libraries/AC_Avoidance/AP_OADijkstra.cpp:        		 "Dijkstra: %s", error_msg);
libraries/AP_AdvancedFailsafe/AP_AdvancedFailsafe.cpp:           "Terminating due to fence breach");
libraries/AP_AdvancedFailsafe/AP_AdvancedFailsafe.cpp:           "Terminating due to RC failure");
libraries/AP_AdvancedFailsafe/AP_AdvancedFailsafe.cpp:           "Terminating due to dual loss");
libraries/AP_AdvancedFailsafe/AP_AdvancedFailsafe.cpp:           "Terminating due to dual loss");
libraries/AP_AdvancedFailsafe/AP_AdvancedFailsafe.cpp:           "Terminating due to range %.1fkm", distance_km);
libraries/AP_Compass/CompassCalibrator.cpp:         		 "Mag(%u) bad orientation: %u/%u %.1f", _compass_idx,
libraries/AP_Compass/CompassCalibrator.cpp:       		 "Mag(%u) internal bad orientation: %u %.1f", _compass_idx, besti, (double)_orientation_confidence);
libraries/AP_Landing/AP_Landing.cpp:        			 "Landing configuration error, invalid LAND_TYPE");
libraries/AP_Landing/AP_Landing_Deepstall.cpp:         		 "Deepstall: Unable to find the elevator channels");
libraries/AP_AdvancedFailsafe/AP_AdvancedFailsafe.cpp:           "Terminating due to dual loss");
libraries/AP_AdvancedFailsafe/AP_AdvancedFailsafe.cpp:           "Terminating due to dual loss");
libraries/AP_AdvancedFailsafe/AP_AdvancedFailsafe.cpp:           "Terminating due to range %.1fkm", distance_km);
libraries/AP_Compass/CompassCalibrator.cpp:                      "Mag(%u) bad orientation: %u/%u %.1f", _compass_idx,
libraries/AP_Compass/CompassCalibrator.cpp:                      "Mag(%u) internal bad orientation: %u %.1f", _compass_idx, besti, (double)_orientation_confidence);
libraries/AP_Landing/AP_Landing.cpp:         			 "Landing configuration error, invalid LAND_TYPE");
libraries/AP_Landing/AP_Landing_Deepstall.cpp:                   "Deepstall: Unable to find the elevator channels");
libraries/AP_Landing/AP_Landing_Deepstall.cpp:                   "Deepstall: Invalid data from AHRS. Holding level");
libraries/AP_Landing/AP_Landing_Slope.cpp:                       "Flare crash detected: speed=%.1f", (double)gps.ground_speed());
libraries/AP_Landing/AP_Landing_Slope.cpp:                       "Landing gear was not deployed");
libraries/AP_Logger/AP_Logger_File.cpp:                          "AP_Logger: stuck thread (%s)", last_io_operation);
libraries/AP_Motors/AP_MotorsHeli.cpp:                           "PreArm: H_RSC_MODE invalid");
libraries/AP_Motors/AP_MotorsHeli.cpp:                           "PreArm: H_RSC_SETPOINT out of range");
libraries/AP_Motors/AP_MotorsHeli.cpp:                           "PreArm: H_RSC_IDLE out of range");
libraries/AP_Motors/AP_MotorsHeli.cpp:                           "PreArm: H_RSC_CRITICAL out of range");
libraries/AP_Motors/AP_MotorsHeli.cpp:                           "PreArm: H_RUNUP_TIME too small");
libraries/AP_Motors/AP_MotorsHeli_Dual.cpp:                      "RSC control mode change failed");
libraries/AP_Motors/AP_MotorsHeli_Dual.cpp:                      "PreArm: H_SW1_H3_PHANG out of range");
libraries/AP_Motors/AP_MotorsHeli_Dual.cpp:                      "PreArm: H_SW2_H3_PHANG out of range");
libraries/AP_Motors/AP_MotorsHeli_Quad.cpp:                      "RSC control mode change failed");
[... cut ...]
libraries/AP_Scripting/lua_scripts.cpp:                          "Lua: %s returned bad result count (%d)", script->name, returned);
libraries/AP_Scripting/lua_scripts.cpp:        		         "Lua: Couldn't allocate a lua state");
libraries/AP_Scripting/lua_scripts.cpp:                          "Scripting: Could not find sandbox");
libraries/AP_Scripting/lua_scripts.cpp:                          "Scripting: Loading sandbox: %s", lua_tostring(L, -1));
libraries/AP_Terrain/AP_Terrain.cpp:                             "Terrain: Allocation failed");

Nadavo6, 2020-09-17:
Hi ,
I am trying to create a complete dictionary of Arducopter Messages and Warnings:
1. Explaining what is the trigger to each one ?
2. Reccomendation as of what is best to do if the message pop up : On-Ground and Airborne.

Thinking it could be a good use to us all...
I could only find a basic explanation of DATA MAVLINK messages ([https://ardupilot.org/copter/docs/logmessages.html].
**Does anyone have an initial list (of messages explained) I can start from?**

Hwurzburg, 2020-09-17:
[gcs messages.txt](https://github.com/ArduPilot/ardupilot_wiki/files/5240563/gcs.messages.txt)
here is a list of gcs messages and which module they are in....daunting task, my friend!

Nadavo6, 2020-09-21:
Thanks Henry !
Me and a couple of guys are working on this, hoping to publish a first draft within a fortnight :)