ArduPilot/ardupilot_wiki #96: Antenna Tracker Wiki needs project information

hamishwillee, 2015-06-15:
The wiki needs more material about the project. Minimally this should include an update to the Project News wiki page and some general information about the project roadmap, members, and how you can join.

I can put this together, but we need to discuss first. Assigned to you Randy

hamishwillee, 2015-09-08:
@rmackay9 Is "additional project information" ever going to happen, or should I just shut this item?

gmorph, 2016-11-15:
There is no roadmap for AntennaTracker nor specific AntennaTracker members or any need to join anything.  I have added a couple of sentences to where uses can find the link to the support forums and where developers can find the link to the gitter channel on the AntennaTracker landing page.
Thanks, Grant.

closed 2016-11-15 by gmorph

hamishwillee, 2016-11-16:
Works for me :-)