ArduPilot/ardupilot_wiki #98: FrSky telemetry: document available and encoded values

hamishwillee, 2015-06-17:
FrSky doesn't use MAVLink so there is some mapping of ArduPilot data to arbitrary fields in the frsky protocol done in the driver. This is useful information to know, and should be added to the [FrSky Telemetry](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-frsky-telemetry/) wiki page.

This issue originated on the forums, and there is [more detail here](http://ardupilot.com/forum/viewtopic.php?f=50&t=12753).

hamishwillee, 2015-06-17:
@badzz I understand you wrote the FrSky driver for ArduPilot? A community member has reverse-engineered some of the protocol [see here](http://ardupilot.com/forum/viewtopic.php?f=50&t=12753) and I am [adding this to the wiki here](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-frsky-telemetry/#protocol_information).

As you can see on the wiki there are some missing FRSKY_ID_ and descriptions. Could you please look at the document and provide the missing IDs and descriptions? I think as the author you will find this fairly trivial :-).

hamishwillee, 2015-08-12:
The provided information [was added](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-frsky-telemetry/#protocol_information). Closing as no more information available at this time.

badzz, 2015-09-26:
Hi,
I am having but it looks fine, i will do a few correction but it is mainly
ok
It was a busy time at that time and I overlooked this email
Sorry

@badzz <https://github.com/badzz> I understand you wrote the FrSky driver
for ArduPilot? A community member has reverse-engineered some of the
protocol see here <http://ardupilot.com/forum/viewtopic.php?f=50&t=12753>
and I am adding this to the wiki here
<http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-frsky-telemetry/#protocol_information>
.

As you can see on the wiki there are some missing FRSKY_ID_ and
descriptions. Could you please look at the document and provide the missing
IDs and descriptions? I think as the author you will find this fairly
trivial :-).

hamishwillee, 2015-09-28:
@badzz Thanks very much - appreciate you're busy and swamped by emails. I've reopened - can you please let me know when you have made the changes.

gmorph, 2016-11-15:
This is still an outstanding issue.  I can see the FrSky ID's still haven't been filled in. Can we get an update on this from @badzz 
Thanks, Grant.

OXINARF, 2016-11-15:
@gmorph Actually @tomlauzon and @floaledm are the new FrSky library maintainers. They said they were going to update the wiki with their new protocol (although that was quite a time ago) so they can probably update this.

tomlauzon, 2016-11-15:
hi guys, sorry, it has been a while. I have to admit I tried to change too many things at the same time and got a bit overwhelmed with editing so many things using rst. Things got in the way but I'll take a look at this and see how to add it all together. This is a good reminder to move my priorities the the frsky page. Thanks

tomlauzon, 2016-11-29:
Haven't forgotten about this. I'm almost done with the update of the FrSky page with the new protocol. Will be including the missing messages for the standard protocol along with the pull request. Looking like a day or two away.

hamishwillee, 2016-11-29:
That's awesome! Thank you.

tomlauzon, 2016-12-02:
I know I said a day or two... Florent wanted to review it one more time so it might be early next week. Turns out users don't need to know the names of the ID variables used inside of ArduPilot, and devs only need to look at the code so the table will be simplified and we'll explain what comes from the FC and what comes from the FrSky sensors so that it's easier for people to understand and configure their transmitter. Thanks for your patience.

m01, 2016-12-03:
Thanks for not forgetting about this 👍
I noticed that the links in the posts at the top of this are out of date, so here are some updated links:
* [FrSky telemetry wiki page](http://ardupilot.org/copter/docs/common-frsky-telemetry.html)
* [forum thread](http://discuss.ardupilot.org/t/frsky-telemetry-document-available-and-encoded-values/5891)

brunoolivieri, 2019-07-12:
Would this Issue still make sense to stay open after the PR been closed?  I believe that could be closed.

rmackay9, 2019-07-12:
I'm not sure about this one..

Hwurzburg, 2019-07-12:
I think it can be closed....

closed 2019-07-12 by rmackay9