ArduPilot/ardupilot_wiki #102: archive / remove 3DR camera control board page

rmackay9, 2015-06-27:
Realistically do we think this board is ever going to materialise?  It would have been nice but leaving it on the wiki is just a distraction from the solutions that actually exist so let's archive it.

http://copter.ardupilot.com/wiki/common-optional-hardware/common-cameras-and-gimbals/common-camera-control-board/

hamishwillee, 2015-06-27:
@CraigElder Thoughts?

zlite, 2015-06-27:
Adding Brandon, since this was his baby. I think we canceled it because the
companion computer (in Solo and sold along with PH2) does most of this for
free.

zlite, 2015-06-29:
Canceled, on account of being behind the times and not worth building, as
the market need for it was small.

We can remove the page on it, if that's the question.

Brandon Basso, PhD :: VP Software Engineering :: 3D Robotics :: Berkeley, CA

CraigElder, 2015-07-01:
There is still some activity regarding the board.  I was contacted by a few
people this spring including one earlier this week.  There are at least two
people that I know who have copied the board and are using it.  I'm sure
Sandro has copies of the instructions but perhaps an archive copy of the
wiki entry could be made as a pull request to
https://github.com/sandrobenigno/3dr-camera-ctrl.  Basically we need a name
change and a different manufacturer for the board

hamishwillee, 2015-07-06:
@CraigElder @rmackay9 So what do I do with this now? Is there a name change we can use/what sort of timeframe are we looking at before one exists. 

Obviously it is easier to find a manufacturer if the board appears to be live and if we make it clear that we are seeking one.

My preference (for now) would be to:
* Rename this to "Open Camera Control Board" (or suggestion for name please ????)
* Change the note up the top of the page to indicate that this board is available for others to manufacturer and pointing to the sandrobenigno github area 
* If board not "owned" in an agreed amount of time we move it to archived. (and if this is OK, what is that timeframe?)

rmackay9, 2015-07-06:
@hamishwillee,@CraigElder,

Not to put too fine a point on it but I think Craig is blocking the removal of this page.

3DR is never going to build it it seems.

I personally don't think a manufacturer is actually going to appear and produce it.  We generally don't have documentation for hardware that does not exist.  The closest thing we have to this is the UAVCAN board page but they do actually exist and there is a manufacturer who is actively looking to produce the board.

I don't know who makes the final decision in these cases where we don't reach a consensus although I think maybe I have final say on what goes onto the copter wiki.

CraigElder, 2015-07-06:
Randy,
Not blocking but it seems a shame to throw away a working project because 3DR has dropped interest in manufacturing the board.  There are other hardware vendors.
I have sent a message to the last person who contacted me about this board to see if they are interested in taking over the project.

hamishwillee, 2015-07-06:
@rmackay9 @CraigElder I see both points of view, which is why the [proposal above](https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/102#issuecomment-118761201) seeks to balance the desire to remove "vapourware" vs the desire to find a manufacturer. 

I propose 6 weeks from today to find a vendor before this is archived. Are we OK with renaming this "Open Camera Control Board" + rest of the proposal?

rmackay9, 2015-07-07:
@hamishwillee, @CraigElder,
I'm happy with the 6 week period to find a manufacturer and the renaming, etc.

hamishwillee, 2015-07-09:
I have [renamed and updated the article note to state we are actively seeking manufacturers](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-cameras-and-gimbals/common-camera-control-board/)

Following discussion with Craig, if we don't find a manufacturer this will move to Dev in 6 weeks.

hamishwillee, 2015-08-12:
What happens with this will affect #73

hamishwillee, 2015-08-18:
@CraigElder We're about 6 weeks from when this was last discussed. Are there committed manufacturers working on this or about to?

hamishwillee, 2015-09-08:
@CraigElder Any more concrete progress on this? I'm not so worried about this now that we have the big note up the top.

CraigElder, 2015-09-14:
Lots of activity but not much progress.  There are problems with the design files that are slowing development but it really comes down to nobody having time or inclination to work on this.

hamishwillee, 2015-09-15:
Up to @rmackay9 then to decide how offended he is by the article. I'm OK with it while there is activity.

rmackay9, 2015-09-15:
I'd interpret the "nobody having time or inclination..." means it's not going to happen.  I say we should remove it because really, it's extremely unlikely it will be picked up.

hamishwillee, 2015-09-16:
OK, propose to archive it for now - will do so on Friday if there are no objections before then.

hamishwillee, 2015-09-24:
I've archived this on all wikis. So it is still available and present but has Archived in the title and is in the Archived section.

closed 2015-09-24 by hamishwillee

rmackay9, 2015-09-24:
thanks @hamishwillee