ArduPilot/ardupilot_wiki #104: make NSH page more visible

rmackay9, 2015-07-01:
I think we should make this page more visible. 
http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/interfacing-with-pixhawk-using-the-nsh/

I think it should be right above "Debugging with GDB".

hamishwillee, 2015-07-06:
Randy, I did this. 

But, do you think we might be better off grouping all this "sort" of information either under [APMProject Developer Information](http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/) (which after all, already has [Interfacing with APM using the Debug Terminal](http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/interfacing-with-apm-using-the-debug-terminal/)?

We might alternatively group all debugging information in a folder "Debugging" (and this could be under [APMProject Developer Information](http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/) if you think that is a good organisation). 

A debugging grouping topic could have:

* [Interfacing with APM using the Debug Terminal](http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/interfacing-with-apm-using-the-debug-terminal/)
* [Interfacing with Pixhawk using the NSH](http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/interfacing-with-pixhawk-using-the-nsh/)
* [Debugging with GDB](http://dev.ardupilot.com/wiki/debugging-with-gdb/)
* [Debugging using JTAG](http://dev.ardupilot.com/wiki/jtag/) - currently not in sidebar!
* [Serial Output](http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/serial_output/) - maybe

Are there any other topics/what do you think?

rmackay9, 2015-07-06:
@hamishwillee, the title "APM Project Developer Information" is not great though because really, the whole dev wiki is project developer info.
If we rename the section to "Debugging" or something like that then grouping all that information under it would be good though.

I think the "Serial Output" and "I2C" pages should be zapped.

hamishwillee, 2015-07-06:
@rmackay9 Agreed that the DevInfo page is not particularly useful and a bit "deceptive". Some of the content is not appropriate under debugging though. I will address this more broadly as #107

closed 2015-07-06 by hamishwillee