ArduPilot/ardupilot_wiki #105: Delete Interfacing with APM using the Debug terminal

rmackay9, 2015-07-01:
http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/interfacing-with-apm-using-the-debug-terminal/

I'm not sure what this page is talking about but it appears to be a precursor to MAVProxy.  I think we can delete this.

We should double check with Tridge first to be sure.

rmackay9, 2015-07-01:
These can go too I think:
http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/serial_output/
http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/i2c/
http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/mavlink-parameters/
http://dev.ardupilot.com/wiki/advanced-configuration-settings/gps_guide/
http://dev.ardupilot.com/wiki/advanced-configuration-settings/gps_input_raw/
http://dev.ardupilot.com/wiki/advanced-configuration-settings/combo_2/
http://dev.ardupilot.com/wiki/advanced-configuration-settings/combo_3/
http://dev.ardupilot.com/wiki/advanced-configuration-settings/combo_4/

and these maybe this page well:
http://dev.ardupilot.com/wiki/advanced-configuration-settings/gps_input/

hamishwillee, 2015-07-09:
Deleted following pages that Grant says can be deleted

* [EZ Set up and Build ArduPilot for PX4 in Windows](http://dev.ardupilot.com/ez-set-up-and-build-ardupilot-for-px4-in-windows)
* [EZ Using Eclipse for ArduPilot Development in Windows](http://dev.ardupilot.com/ez-using-eclipse-for-ardupilot-development-in-windows)
* [EZ Building APM2.x with Make in Windows](http://dev.ardupilot.com/ez-building-apm2-with-make-in-windows)
* [EZ Developer Walk Throughs for Windows](http://dev.ardupilot.com/ez-developer-walk-throughs-for-windows)
* [EZ Get Arduino and Initialize it to work with ArduPilot](http://dev.ardupilot.com/wiki/ez-developer-walk-throughs-for-windows/)
* [EZ Arduino Compile and Upload the ArduPilot Firmware to your APM](http://dev.ardupilot.com/ez-arduino-compile-and-upload-the-ardupilot-firmware-to-your-apm)
* [EZ Install the ArduPilot Source Code on your Computer Using Zip](http://dev.ardupilot.com/ez-install-the-ardupilot-source-code-on-your-computer-using-zip)
* [Programming the Atmega328p using AVR Studio](http://dev.ardupilot.com/wiki/avr_studio/)
* [Building the code for APM2.x on Windows with Arduino](http://dev.ardupilot.com/wiki/building-apm2xe-on-windowsdev_for_windows/) (old version)
* [Code Overview (Plane)](http://dev.ardupilot.com/wiki/arduplane-code-overview/)
* [ArduPilot Arduino Windows Setup and Compile Walk Through](http://dev.ardupilot.com/ardupilot-arduino-windows-setup-and-compile-walk-through)

hamishwillee, 2015-07-09:
@rmackay9 - all of the above are deleted. Also checked with tridge and http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/interfacing-with-apm-using-the-debug-terminal/ is (as you thought) something that can be deleted (and was).

This issue is therefore complete, and I'm closing it.

closed 2015-07-09 by hamishwillee