ArduPilot/ardupilot_wiki #106: Add Solo developer page to menu and landing page under a new section "RTF vehicles"

rmackay9, 2015-07-02:
To help developers work with the Solo I've started on a new wiki page for the developer site.
http://dev.ardupilot.com/wiki/solo/

Hamish, could you add this to the menu for me and also create a new section on the landing page (http://dev.ardupilot.com/) called "RTF vehicles".  For now it'll just have one link underneath pointing to this new page.  Hopefully in the future we will get some more pages for Walkera vehicles and other DroneCode members.
I think this section should be just above "How the team works".

hamishwillee, 2015-07-06:
Done.

I have subedited the page and grouped it in the sidebar under [RTF Vehicle Developer Information](http://dev.ardupilot.com/wiki/ready-to-fly-rtf-vehicle-developer-information/)

hamishwillee, 2015-07-06:
Updated the grouping topic with "information about what sort of pages should go in here". IMO this should probably say a little about using companion computers on Solo? Closing for now, but recommend you add @rmackay9

closed 2015-07-06 by hamishwillee

rmackay9, 2015-07-06:
@hamishwillee, yes we need to build out the Solo page a fair bit more.
What's there is the current limit of my knowledge on working with the Solo I'm afraid.  As I and others learn more about what's on the companion computers hopefully we can build it out.

hamishwillee, 2015-07-06:
Sounds great to me - at least there is somewhere on the wiki for this information to live now.