ArduPilot/ardupilot_wiki #107: Restructure Dev Wiki top level topics

hamishwillee, 2015-07-06:
The dev wiki top level structure is confusing. It has top level grouping topics like "APM Project Developer Information" when that is the content of the whole wiki (and which is also a content topic related to vagrant).

The aim of this is to split out and group topics related to to the main things that are interesting to developers: testing, contributing, building, developing, debugging etc.

hamishwillee, 2015-07-06:
I have created the following grouping topics and populated appropriately.

* [Contributing](http://dev.ardupilot.com/wiki/debugging/)
* [Debugging](http://dev.ardupilot.com/wiki/debugging/)
* [MAVLink Commands](http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/mavlink-commands/) is now a parent document for the other docs. It includes links to command docs on other wikis too.
* [Code Editing Tools and IDEs](http://dev.ardupilot.com/wiki/code-editing-tools-and-ides/)
* [Appendix](http://dev.ardupilot.com/wiki/appendix/) 

I deleted:

* [I2C Sensors](http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/i2c/) - as you say, not particularly "developer relevant"
*  "Serial Output" - you indicated not relevant for a number of years.
* "APM Project Developer Information" - deceptive grouping topic. Based on content could have been "Setting up vagrant build" but we have a better topic for that now. All contents moved out.

gmorph, 2016-11-15:
I think the top level now looks fine - you ok to close this Hamish?

hamishwillee, 2016-11-15:
Thanks @gmorph - yes, I did this (just needed some sign-off). It is definitely improved, so lets close (there are other issues, but they are more related to new build system not being documented)

closed 2016-11-15 by hamishwillee