ArduPilot/ardupilot_wiki #111: Rover: Update entry points to refer to Pixhawk

hamishwillee, 2015-07-27:
In particular http://rover.ardupilot.com/wiki/apmrover-setup/ , but the whole of the initial material needs to be reveiwed/reworked.

gmorph, 2015-08-03:
Its been updated.  Please review and if happy close this out.

hamishwillee, 2015-08-04:
Hi @gmorph 

Great job. This maps the similar structure in Copter/Plane. As with those wikis the APM/PX4 structure does not match the Pixhawk structure, but at least we have a good framework for how we build autopilot sections going forward.

What I have done:

1. I updated the [top level page](http://rover.ardupilot.com/wiki/apmrover-setup/) ([diff](http://rover.ardupilot.com/wp-admin/revision.php?from=4492&to=4498)) to change title from "Rover Setup with APM 2.x" to "Rover Setup and Configuration" - ie remove the APM specifier in the title.

  * I don't really like "First Time" for title, but you could change if you like.

2. [Rover Code Configuration](http://rover.ardupilot.com/wiki/rover-code-configuration/) ([diff](http://rover.ardupilot.com/wp-admin/revision.php?from=4493&to=4499))

  * Changed the stub/url to remove that annoying -2 on the end of the URL
  * TIdied the "Load the code" section to instead point to the common instructions
  * General subedit

3. [Installing Mission Planner](http://rover.ardupilot.com/wiki/common-install-mission-planner/) ([diff](http://rover.ardupilot.com/wp-admin/revision.php?revision=4500)) - fixed image/layout rendering issues and simplified instructions

4. Added these Pixhawk sub topics to the menu:

  * [Powering the Pixhawk](http://rover.ardupilot.com/wiki/common-powering-the-pixhawk/)
  * [Compatible RC Transmitter and Receiver Systems (Pixhawk/PX4)](http://rover.ardupilot.com/wiki/common-pixhawk-and-px4-compatible-rc-transmitter-and-receiver-systems/)

5. Renamed this to [Rover PX4 Wiring and Quick Start](http://rover.ardupilot.com/wiki/apmrover-setup/px4-setup/) for consistency.

Questions:

1. "Reassigning your RC transmitter stick channels" is present in [Rover APM2.x Wiring and Quick Start](http://rover.ardupilot.com/wiki/rover-apm2-setup/#reassigning_your_rc_transmitter_stick_channels) and [Rover PX4 Wiring and Quick Start](http://rover.ardupilot.com/wiki/apmrover-setup/px4-setup/#reassigning_your_rc_transmitter_stick_channels).

  * Should this be in its own section shared by all the autopilot pages? I was thinking this was [Configuration](http://rover.ardupilot.com/wiki/rover-code-configuration/)

* Do we want to add a [Detailed Vehicle Builds](http://copter.ardupilot.com/wiki/common-common-airframe-builds/) section (like in Copter)? If so we could seed it with this section http://rover.ardupilot.com/wiki/apmrover-setup/px4-setup/#px4_rover_build_with_brushless_traxxas_stampede_4wd_truck

3. For Copter we have a separate topic for [connecting escs and motors](http://copter.ardupilot.com/wiki/initial-setup/assembly-instructions/connect-escs-and-motors-pixhawk/). Is this something that needs more detail than we already have in the quickstart?

4. I did not change the "pins" information for connecting steering - I am hoping you will check the discussion and make changes if appropriate.

gmorph, 2015-08-04:
Answers to Questions:-)

1.    "Reassigning your RC transmitter stick channels" is present in Rover APM2.x Wiring and Quick Start and Rover PX4 Wiring and Quick Start.
        Should this be in its own section shared by all the autopilot pages? I was thinking this was Configuration

Rover and Plane do it the same way but Copter is different.  We could have a common page for Rover and Plane?  Copter I believe will change over at some stage in the future.

    Do we want to add a Detailed Vehicle Builds section (like in Copter)? If so we could seed it with this section. I think a similar Pixhawk build would be REALLY good. Can you do that?

Yes - I reckon that's a good idea.  I'll create a task and assign it to myself.

    For Copter we have a separate topic for connecting escs and motors. Is this something that needs more detail than we already have in the quickstart?

Yes it does - more so for Rover as those ESCs have braking.  But I don't have any more details to add at the moment - I'm not an expert.  I think this should be an issue item but not assigned to anyone.  I'll raise that too.

    I did not change the "pins" information for connecting steering - I am hoping you will check the discussion and make changes if appropriate.

DONE.

    Would be great to have a more detailed wiring diagram for rover/Pixhawk - a bit like this one http://copter.ardupilot.com/wiki/advanced-pixhawk-quadcopter-wiring-chart/ but for rover.

That is a gorgeous picture but I don't have the skill to produce one of those.  Can you?

hamishwillee, 2015-08-05:
@gmorph Thanks very much. As you can see I've linked the "spawned" issues for all of the questions. 

I'll therefore close this isse.

closed 2015-08-05 by hamishwillee