ArduPilot/ardupilot_wiki #113: Add wifi and telephone network (aka 3G, 4G) telemetry and video wiki pages

rmackay9, 2015-07-31:
One example for wifi telemetry is here:
http://diydrones.com/profiles/blogs/wifi-telemetry-with-cheap-3-esp8266?xg_source=activity

For the wifi video there's the wifi broadcast stuff:
http://diydrones.com/forum/topics/3-km-hd-fpv-system-using-commodity-hardware

and the navio+ guys have something too.
http://docs.emlid.com/Navio-APM/video-streaming/

Tower guys say they're adding video input support as well so that might be a useful way for people to display the video.  MP already supports it.

grcasanova, 2018-05-17:
Hi, I have the same need, for video I will use uv4l drivers and a Raspberry compatible camera to leverage GPU power and avoid overloading the CPU, video stream would be directly on the browser as it initiates a local web server on the pi.
On the telemetry, I would use connectionless sockets for speed, I already have a PoC, it could be integrated in Ardupilot while GCS side a classical USB 433MHz telemetry device input could be emulated

rmackay9, 2018-05-17:
[APSync](http://ardupilot.org/dev/docs/apsync-intro.html) and [Maverick](https://goodrobots.github.io/maverick/#/) are the best open source solutions we have at the moment for live video.

grcasanova, 2018-05-17:
Do they run on 3G? On their websites it seems they're limited to WiFi

rmackay9, 2018-05-17:
@grcasanova, wifi only I'm afraid for now

grcasanova, 2018-05-17:
In my case I have a fixed wing aircraft so I will have to run everything via mobile network. Anyway do you have any resource on how to get telemetry information via MAVlink? So that I can parse incoming data and send it back to the GCS.

hamishwillee, 2018-05-18:
@grcasanova YOu might want to use http://python.dronekit.io/about/index.html for getting the telemetry info.

grcasanova, 2018-05-18:
maybe something in C/C++ too? As for the sockets I am using a C library and would rather stick to it without having external bindings. Thanks

brunoolivieri, 2019-07-12:
Would this Issue still make sense to stay open?

rmackay9, 2019-07-12:
yes, we should keep this open.  There is definitely a 3G/4G telemetry solution discussed in the Rover forums somewhere (sorry, don't have a link).  Many users want to do this so we should provide some details on how it can be done.

Hwurzburg, 2019-07-12:
Fyi, here is a link to some how-to stuff I did in RCG: https://www.rcgroups.com/forums/showthread.php?3157728-Cellular-Telemetry-and-Control-in-Ardupilot...... Andruav is semi-maintained but still active ....
however, UavCast is very supported, works well (cell telem only works below 200-300ft altitude..cell antennas are very directive...not many cell phones above 300feet AGL), and I did a how-to guide: https://www.dropbox.com/s/xpm85d640e89tf6/Cellular%20Telemetry%20HowTo.pdf?dl=0
maybe I should convert to Wiki format?

rmackay9, 2019-07-12:
Ah, yes, it's UAVCast that I was thinking of.  I plan to give it a try at some point but I think we should add it to our wiki some time soon.  Even just a link to their website would be a good start.

auturgy, 2019-07-13:
It’s been in the wiki for a while 

Regards,

James

rmackay9, 2019-07-13:
ok, closing this issue now

closed 2019-07-13 by rmackay9