ArduPilot/ardupilot_wiki #114: Update CHDK documents with new circuit if appropriate

hamishwillee, 2015-08-04:
See http://diydrones.com/forum/topics/using-aux-pins-as-relays-for-chdk?commentId=705844%3AComment%3A2062362&xg_source=msg_com_forum

This shows new circuit for cable. Evaluate how useful this is and add any needed diagrams to CHDK docs. If this is good, it is an enhancement to [this section](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-cameras-and-gimbals/common-apm-to-chdk-camera-link-tutorial/#cable_for_connecting_to_pixhawk)

gmorph, 2016-11-15:
I don't think we need that circuit diagram as the one we have  in here
http://ardupilot.org/rover/docs/common-apm-to-chdk-camera-link-tutorial.html?highlight=chdk
provides a better level shifting circuit and was in the original discussion.
However, we have plenty of APM2 configuration information but not enough pixhawk configuration information for CHDK and instead we reference this page
http://www.tuffwing.com/support/pixhawk_camera_trigger_cable.html
in our wiki but we should incorporate that information into the wiki itself without having to rely on an external source.

brunoolivieri, 2019-07-12:
Would this Issue still make sense to stay open?  I believe that could be closed.

rmackay9, 2019-07-12:
I suspect this should stay open.  I've never looked in depth at our wiki pages for triggering a camera but I suspect they need work.

Hwurzburg, 2026-09-29:
Closing: the CHDK pages (common-apm-to-chdk-camera-link-tutorial and common-chdk-camera-control-tutorial) were archived as obsolete in 2024. Setting up camera triggering on current firmware is covered in [Camera Shutter Configuration](https://ardupilot.org/copter/docs/common-camera-shutter-with-servo.html). Please reopen if you would like the CHDK pages brought back and updated for current firmware.

closed 2026-09-29 by Hwurzburg