ArduPilot/ardupilot_wiki #115: Add a Detailed Vehicle Builds section

gmorph, 2015-08-04:
Use the vehicles in the getting started section to seed this page.

hamishwillee, 2015-08-05:
I made the following changes:

1. Added [Detailed Vehicle Builds](http://rover.ardupilot.com/wiki/common-common-airframe-builds/) to the menu. Updated this page to refer to "common frames" rather than "common airframes".
2. Split out [PX4 Rover Build with Brushless Traxxas Stampede 4WD Truck](http://rover.ardupilot.com/wiki/px4-rover-build-with-brushless-traxxas-stampede-4wd-truck/) from the quickstart and subedited. Added to the menu.
3. Minor subedit [Rover PX4 Wiring and Quick Start](http://rover.ardupilot.com/wiki/rover-px4-quickstart/) after splitting out above page (and linking to it)

@gmorph Can you please do quick sanity check. Also are there any other vehicle builds we can split out?

Otherwise I think this can be closed?

gmorph, 2015-08-05:
Reviewed and it looks great.  It will be very good if we can add more
vehicles to that page.  I think the equivalent for Plane and Copter would
be great.

Thanks, Grant.

hamishwillee, 2015-08-06:
> I think the equivalent for Plane and Copter would be great.

Way ahead of you - equivalent sections linked below. Plane only has the Bixler, while Copter has a few builds and some generic build information that doesn't quite fit anywhere else. As you say, extending both Rover and Plane with more real example builds would be great.

* [Plane](http://plane.ardupilot.com/wiki/common-common-airframe-builds/)
* [Copter](http://copter.ardupilot.com/wiki/common-common-airframe-builds/)

Closing this as complete.

closed 2015-08-06 by hamishwillee