ArduPilot/ardupilot_wiki #118: Rover Detailed Wiring Diagram (Pixhawk)

hamishwillee, 2015-08-05:
We have very basic Rover wiring diagrams mostly for the connections [to the autopilot board](http://rover.ardupilot.com/wiki/common-pixhawk-wiring-and-quick-start/#pixhawk_wiring_chart). 

It would be good to have a full diagram, showing all the connections for a typical build, including the controller board essential connections (power, safety switches, receiver) and optional hardware - telemetry, sonar etc etc.

A good example of the sort of thing we'd like is Copter's [Advanced Quadcopter Wiring Chart](http://copter.ardupilot.com/wiki/advanced-pixhawk-quadcopter-wiring-chart/).

gmorph, 2016-11-15:
I can do this.

brunoolivieri, 2019-07-12:
I think that is cover by page http://ardupilot.org/rover/docs/rover-motor-and-servo-connections.html

Would this issue still make sense to stay open?

rmackay9, 2019-07-12:
yes, i think we can close this now, txs

closed 2019-07-12 by rmackay9