ArduPilot/ardupilot_wiki #119: Reassigning RC transmitter stick channels

hamishwillee, 2015-08-05:
"Reassigning your RC transmitter stick channels" is present in [Rover APM2.x Wiring and Quick Start](http://rover.ardupilot.com/wiki/rover-apm2-setup/#reassigning_your_rc_transmitter_stick_channels) and [Rover PX4 Wiring and Quick Start](http://rover.ardupilot.com/wiki/apmrover-setup/px4-setup/#reassigning_your_rc_transmitter_stick_channels).

Should this be in its own section shared by all the autopilot pages? I was thinking this was [Configuration](http://rover.ardupilot.com/wiki/rover-code-configuration/)

This discussion spawned off #111

hamishwillee, 2015-08-05:
From @gmorph 

>Rover and Plane do it the same way but Copter is different. We could have a common page for Rover and Plane? Copter I believe will change over at some stage in the future.

hamishwillee, 2015-08-05:
@gmorph Is it correct that this is "optional setup" - ie you can and probably should just accept default configuration? In that case I think a separate common page makes sense. 

I can't see any information on this for Copter stick setup - is there any existing information you know of?

rmackay9, 2015-08-05:
I think for the bottom 4 channels all vehicles are the same as they all use RCMAP.  I created a page for this for copter at least:
http://copter.ardupilot.com/wiki/common-rcmap/

gmorph, 2015-08-05:
Thanks for clarify Randy.

Yes it is optional as you can change the RCMAP.  You can probably achieve
what your after with vehicle specific sections on the page Randy created.

Thanks, Grant.

hamishwillee, 2015-08-06:
Thanks @rmackay9 @gmorph. 

* I modified [Rover APM2.x Wiring and Quick Start](http://rover.ardupilot.com/wiki/rover-apm2-setup/#reassigning_your_rc_transmitter_stick_channels) and [Rover PX4 Wiring and Quick Start](http://rover.ardupilot.com/wiki/rover-px4-quickstart/#reassigning_your_rc_transmitter_stick_channels). to say: 

  > The default transmitter stick configuration should be suitable in almost all cases. If you do need to change them, then see*RCMAP Input Channel Mapping* <link>.

* I updated [RCMAP Input Channel Mapping](http://planner.ardupilot.com/wiki/common-rcmap/) ([diff](http://planner.ardupilot.com/wp-admin/revision.php?from=4633&to=4732)) to add a a section on Rover notes, visible only if you're on Rover. This includes the information that you can ignore the yaw and pitch settings, and also information about the LEARN_CH, MODE_CH (not quite the right place, but not a bad one!).

I think all the other information in the above links was redundant (instructions on how to connect to the flight controller and connect to mission planner).

Closing as done - please review the RCMAP topic and reopen this if there is something you don't like.

closed 2015-08-06 by hamishwillee