ArduPilot/ardupilot_wiki #120: Plane: Improved flaperon as spoilers documentation

hamishwillee, 2015-08-05:
The problem/setup is as described here: https://github.com/diydrones/ardupilot/issues/2524

@marcmerlin has offered to write the documentation.

hamishwillee, 2015-08-05:
@marcmerlin If you want to join the proper wiki and edit directly, please message me at "hamish at 3drobotics dot com" using your preferred email account.

It might just be easier for you to write the document on this repo's [wiki](https://github.com/diydrones/ardupilot-wiki-issue-tracker/wiki) and link your new pate to this issue. I can then copy the content across for you. 

Can you ETA when you might have a chance to write this?

marcmerlin, 2015-08-05:
@hamishwillee is "I just spent 2h writing and testing it and I'm done" soon enough? :)

You can steal the content from
﻿﻿http://marc.merlins.org/perso/rc/post_2015-08-04_Using-Flaperons-With-Ardupilot.html

marcmerlin, 2015-08-05:
BTW, you don't want the title to say "spoilers", spoilers are when the flaps go up,killing lift from the wing and here we want them to go down to allow having extra lift and not stall at lower speeds.
Ideally APM::Plane would allow flaperon support to do flaps or spoilers, but it does not.
To be fair, I've never needed spoilers on a plane without flaps + ailerons. If I have flaps + ailerons, crow flaps are helpful for landing though (typically a big glider with lots of lift).

hamishwillee, 2015-08-05:
Thanks very much Marc! Appreciated the content - you know far more about Plane than me. I have copied this in "verbatim", and now I'm giving it a subedit: [Flaperons on Plane](http://plane.ardupilot.com/wiki/flaperons-on-plane/). When its done, I'll get @gmorph to review and we'll add it to the sidebar.

Just FYI, your blog title should be "Marc's Public Blog - Remote Control Aircraft" (omit the last "s") for correct English.

marcmerlin, 2015-08-05:
On Tue, Aug 04, 2015 at 10:04:38PM -0700, Hamish Willee wrote:
> Thanks very much Marc! Appreciated the content - you know far more about Plane than me. I have copied this in "verbatim", and now I'm giving it a subedit: [Flaperons on [...]
 
Thanks. Yes, I'd like someone to review this for correctness.

> Just FYI, your blog title should be "Marc's Public Blog - Remote Control Aircraft" (omit the last "s") for correct English.

You are absolutely correct, and I can't belive I never noticed this
typo. As a pilot, I should know better :-/

Fixed :)

Marc

KIrill-ka, 2015-08-05:
Thank you Marc! I'd like to add some details on TRIM/MIN/MAX usage.
1) MIN and MAX for flaperon output channels limit deflection of flaperons.
2) TRIMs for flaperon output channels allow to set neutral position of individual flaperons
3) RC1_TRIM acts as normal aileron trim, RC1_MIN and RC1_MAX should match the TX setting.
4) If you assign flap input and output function to one channel (RC5), adjust TX limits to match MIN and MAX defined at step 1) for RC5.

marcmerlin, 2015-08-05:
@KIrill-ka that sounds like a good addition.
Do you have access to http://plane.ardupilot.com/wiki/flaperons-on-plane/ yourself? (because I don't).

I could update my blog post but of course it would be better if the ardupilot page was the up to date version with all the details (and I'm going to put a pointer from my page to there)

marcmerlin, 2015-08-05:
Mmmh, one typo to fix (sorry):
s/There is only including documentation/There is only incomplete documentation/

KIrill-ka, 2015-08-05:
> Do you have access to http://plane.ardupilot.com/wiki/flaperons-on-plane/ yourself? (because I don't).

No, I don't. It takes me a lot of time to write in English...

marcmerlin, 2015-08-05:
@hamishwillee 
I've updated http://marc.merlins.org/perso/rc/post_2015-08-04_Using-Flaperons-With-Ardupilot.html as per suggestions from @KIrill-ka 
I've only changed the notes section at the bottom.

hamishwillee, 2015-08-05:
Thanks very much @marcmerlin @KIrill-ka . I have added the notes to the end of my version. Still working on it.

hamishwillee, 2015-08-06:
On hold while @gmorph does some testing. This will be simplified because the blog is about the "journey/problem", while wiki pages only need to show "the right way to set up flaperons"

marcmerlin, 2015-08-06:
On Wed, Aug 05, 2015 at 05:00:48PM -0700, Hamish Willee wrote:
> On hold while @gmorph does some testing. This will be simplified because the blog is about the "journey/problem", while wiki pages only need to show "the right way to set up flaperons"

Totally, the way I wrote it was meant for a blog, but indeed for a howto
page, some sentences need to be reworked or deleted.

Marc
-- 
"A mouse is a device used to point at the xterm you want to type in" - A.S.R.
Microsoft is to operating systems ....
                                      .... what McDonalds is to gourmet cooking
Home page: http://marc.merlins.org/

hamishwillee, 2015-08-07:
The document is now updated and complete - http://plane.ardupilot.com/wiki/flaperons-on-plane/ - an excellent test and update by @gmorph. Thanks!

Thankyou @marcmerlin for your contribution. Grant tells me it pretty much just worked. 

Closing this now.

closed 2015-08-07 by hamishwillee