ArduPilot/ardupilot_wiki #121: move GPS configuration to Advanced configuration section

rmackay9, 2015-08-05:
I think we should move this GPS configuration page to the Copter's Advanced configuration section.
http://copter.ardupilot.com/wiki/common-optional-hardware/common-positioning-landing-page/common-installing-3dr-ublox-gps-compass-module/common-ublox-gps/

hamishwillee, 2015-08-05:
@rmackay9  The plan so far has been to keep optional hardware docs together. I would expect this in "Advanced Configuration" if it were a generic form of configuration (ie gps product agnostic) or if every Copter setup had to use UBlox.

I'm not opposed, but I do want to understand why it is better there, to help me refine our overall structure.

rmackay9, 2015-08-05:
I think pretty much every ardupilot vehicle uses a Ublox and configuring the GPS is something that we wish people didn't have to do but it seems they sometimes do especially if they're using the newer M8 Ublox which is not sold by 3DR and thus comes with god-knows-what standard config.

I think practically speaking the GPS is not optional hardware, but a 2nd GPS probably is.
It's possible to fly ardupilot without a GPS but it's got about 10% of it's total functionality.

hamishwillee, 2015-08-05:
Thanks. I've moved [UBlox GPS Configuration(]http://copter.ardupilot.com/wiki/common-optional-hardware/common-positioning-landing-page/common-installing-3dr-ublox-gps-compass-module/common-ublox-gps/) into Advanced Configuration on copter wiki as you suggested. Should it be renamed?

As part of this I tidied [3DR UBlox GPS + Compass Module](http://copter.ardupilot.com/wiki/common-optional-hardware/common-positioning-landing-page/common-installing-3dr-ublox-gps-compass-module/) and globally removed a separate common topic on UBlox LEDs, which is duplicated by information in this topic.

hamishwillee, 2015-08-05:
@gmorph Randy wanted to move  [UBlox GPS Configuration](http://copter.ardupilot.com/wiki/common-optional-hardware/common-positioning-landing-page/common-installing-3dr-ublox-gps-compass-module/common-ublox-gps/) into Advanced Configuration on copter wiki - on basis that most people use this compass, it is not always configured properly, and they are more likely to find it there. 

Question - would you like to have it there on plane/rover too?

gmorph, 2015-08-05:
Yes please move it.  It is most definitely an advanced topic.

Thanks, Grant.

hamishwillee, 2015-08-05:
They have been moved. I think this is now complete. Closing.

closed 2015-08-05 by hamishwillee