ArduPilot/ardupilot_wiki #122: Add doc for ArduCopter-ROS/Gazebo SITL

hamishwillee, 2015-08-05:
Originate in discussion on [drones-discuss here](https://groups.google.com/forum/#!msg/drones-discuss/sgGNg7IpqN8/QUCknBuxBAAJ). There is an associated PR: https://github.com/diydrones/ardupilot/pull/2660

@alexbuyval has offered to write the page

alexbuyval, 2015-08-15:
Hi,
I have finished a draft version http://dev.ardupilot.com/?post_type=wiki&p=5988&preview=true

Please note, English is not native language for me. So it would be great if somebody checks my English grammar.

Also I have a question. I have used a link to master branch of ardupilot on my wiki-page. However, there is no my PR in master branch yet. 
What should I do now? Should I wait the PR or should I change the link to my fork?

Best Regards,
Alex

hamishwillee, 2015-08-17:
Hi Alex

Thanks very much for this. 

> What should I do now? Should I wait the PR or should I change the link to my fork?

I suggest you make the link to your fork, and we can change later.

> So it would be great if somebody checks my English grammar.

The overview still needs work (discussed below) but I subedited the rest  and it wasn't bad at all! 

1. "simple control node" link broken

2. The overview is a bit confusing. It is not really clear what this delivers. 

Some questions below:

> Now there is only one approach to test the Copter with SITL by using Mavproxy. However this approach is not good for testing of autonomous flights particularly for indoor flights. 

Why? MAVProxy can run missions. Also you aren't forced to use MAVProxy - you can connect to SITL normally using another GCS like Mission planner.

Can you explain the shortcomings of MAVProxy and why Gazebo is better?

> Another side the ROS and Gazebo are famous framework and simulator in the robotic field. The ROS framework contains many packets of robotic software for different task. Also there are several copter models and plugins in Gazebo, which we can use for simulation. You can see common architecture of this approach on picture below. 

So I think this is your main point. I think what you are saying firstly  is that Gazebo/SITL provide a better simulation than the basic copter simulation (ie with several different models). 

Secondly you are saying that there are many "tools" (Called packets in ROS-speak) for different tasks, so you can do more than MAVProxy in terms of sending automated commands "out of the box". Is that correct?

To put it differently, normally we have a copter model in SITL and we drive SITL by sending it mavlink commands from MAVProxy. What you've done is replaced the bit of SITL which models the copter, and also replaced MAVProxy so that now your control messages come from ROS. Is that correct?

Would it be possible to just change the Copter model to gazebo but still drive the vehicle/simulator using MAVProxy?

3. Can you change the copter model used? Can you change to a plane model. Is there anything we should say about configuring the model further?

4. In the line:

> Some topics can be used to control the copter – e.g. mavros/rc/override/.

Would be good to say what this topic does to control the copter. ie what is it for?

5.  don't understand this note - it isn't clear why you need to modify the mavteleop or what it does for you.

> Note:By default I run ‘mavteleop’ node which publishes control messages into ‘mavros/rc/override_joy/’. So if you are going to use only a joystick control, please replace this topic on ‘mavros/rc/override/’ in mavteleop script

6. YOu say "Then you should arm the copter and set ‘ALT_HOLD’ mode by joystick. Then you can set a desired altitude by joystick."- does that mean you can toggle some button on the joystick to set alt-hold mode?

Thanks for this!

hamishwillee, 2015-08-17:
P.S. Once we've got this to the point where I get it we'll publish it for broader review.

alexbuyval, 2015-08-18:
Hi Hamish,

Thank you for review and corrections!

>I suggest you make the link to your fork, and we can change later.

I have changed the link to my fork.

>Why? MAVProxy can run missions. Also you aren't forced to use MAVProxy - you can connect to SITL normally using another GCS like Mission planner.
Can you explain the shortcomings of MAVProxy and why Gazebo is better?

I have added the following:
*For instance, you cannot describe indoor environment (walls, doors and so on) or cannot add some additional sensors into the copter model. Also you cannot run own control algorithm for testing autonomous flight.*

>Secondly you are saying that there are many "tools" (Called packets in ROS-speak) for different tasks, so you can do more than MAVProxy in terms of sending automated commands "out of the box". Is that correct?

Yes, it is correct.

>To put it differently, normally we have a copter model in SITL and we drive SITL by sending it mavlink commands from MAVProxy. What you've done is replaced the bit of SITL which models the copter, and also replaced MAVProxy so that now your control messages come from ROS. Is that correct?

Yes, it is correct too.

>Would it be possible to just change the Copter model to gazebo but still drive the vehicle/simulator using MAVProxy?

Yes, it is possible, but it does not make sense for ROS developers.

>Would be good to say what this topic does to control the copter. ie what is it for?

I have added the following:
*Using this topic you can send a command to copter which override regular RC control. See mavros wiki for details.*

>Can you change the copter model used? Can you change to a plane model. Is there anything we should say about configuring the model further?

I have added the following:
*Using this approach you can easy change the copter model in Gazebo. For instance, there are 6 ready URDF models in 'rotors_simulator' package.*

>don't understand this note - it isn't clear why you need to modify the mavteleop or what it does for you.

I have added the following:
*...(not mavros/rc/override/). I am using it to combine the joystick control with automatic control.*

>YOu say "Then you should arm the copter and set ‘ALT_HOLD’ mode by joystick. Then you can set a desired altitude by joystick."- does that mean you can toggle some button on the joystick to set alt-hold mode?

I have added the folowing:
*On Saitek ST290Pro model of  joystick you should push '4' button to set 'ALT_HOLD' mode and '2' button to arm the copter.*

Best Regards,
Alex

hamishwillee, 2015-08-19:
Thanks Alex - very useful to know. I've updated the [overview](http://dev.ardupilot.com/wiki/using-rosgazego-simulator-with-sitl/#overview) to make it clearer and done some other minor changes. I've set it to public but not added it to the sidebar yet. I'll email the wiki editors group to get broader feedback now.

alexbuyval, 2015-08-19:
Thank you, Hamish.

hamishwillee, 2015-08-24:
No response from Wiki editors yet. Never-the-less this is IMO good so I have added it to the sidebar. Note, changed URL http://dev.ardupilot.com/wiki/using-rosgazebo-simulator-with-sitl/

I'll mark this closed since it is now fully published.

hamishwillee, 2015-08-24:
... Thanks @alexbuyval

closed 2015-08-24 by hamishwillee