ArduPilot/ardupilot_wiki #126: Add FrSky SmartLink converter to FrSky telementry wiki page

rmackay9, 2015-08-07:
https://www.airborneprojects.com/product/apm-mavlink-to-frsky-smartport-converter/

mentioned at the bottom of this wiki page: https://github.com/diydrones/ardupilot/issues/1587#issuecomment-128342271

hamishwillee, 2015-08-07:
Done. 

Can you check it please [FrSky Telemetry](http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-frsky-telemetry/) ([diff](http://planner.ardupilot.com/wp-admin/revision.php?from=4646&to=4745))

I also deleted the redundant article [How to Get APM/Pixhawk telemetry on your Taranis](http://copter.ardupilot.com/wiki/common-how-to-get-apmpixhawk-telemetry-on-your-taranis/) - as discussed.

lvale, 2015-08-07:
@hamishwillee @rmackay9 That is a copy from this project http://diydrones.com/forum/topics/amp-to-frsky-x8r-sport-converter and its derivatives.

ptsneves, 2015-08-07:
@lvale Hi! I would like to point out why I think it is not a copy.

The code is based on Scott Fly's work which is based Rolfs work but surely not a copy. The contribution is also attributed. If you look at the code it is not even normal Arduinos *.ino files but Makefile based. This eliminates the need for global variables typical of the Arduino environment and allows true object code separation. The code is also much more maintainable and readable in my opinion.

Another aspect is that the versions being sold are for Teensies and what we are proposing is an Arduino Nano which is not even the same microchip and a much cheaper alternative. Also our Arduino Nano is adapted with an LDO that enables it to work with any RC voltage off the shelve. For the Teensy that is a given, but for the Arduino Nano it is not, and DIYers will find it much cheaper than the alternative. Also the Teensy is not directly supported by open source Linux distributions like Debian.

This version was developed in-house and launched yesterday so for sure nobody is selling this yet.

Regarding the OpenTX versions there should definitely be a warning about it and we will also update our manual to reflect it. 

The Ardupilot version we used is the one based on the NAVIO fork and we did not experience any difficulties because we only issue MAVLink data requests and do not rely on the direct output of FrSkySmartPort from the telemetry ports.

I am open to further clarifications if needed.

lvale, 2015-08-07:
@ptsneves Not to start a discussion, but I added the expression "derivatives"....

rmackay9, 2015-08-08:
@lvale, very happy to add extra links to other alternatives.  We could group this particular solution and all other equivalent solutions onto this one wiki page.

hamishwillee, 2015-08-08:
@lvale @rmackay9 hence the line in the article: Please feel free to add your own solutions (or ask us to).

Clooney82, 2015-09-28:
Maybe you can add also this solution to the wiki page.
 https://github.com/Clooney82/MavLink_FrSkySPort

hamishwillee, 2015-09-29:
@lvale @rmackay9 Any thoughts on how https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/126#issuecomment-143773845 might fit in? It looks to me like new code for a Teensy based solution - but perhaps not "solid" enough to justify adding?

lvale, 2015-09-29:
I'm also involved on that project :) so my comments might be "somewhat" biased, but IMHO it's currently the most versatile solution of having "decent" feedback via telemetry on FrSky. It is also the base of the projects that are linked commercially....
As being "solid" my project was a "forefather" of that and is being used a lot even before native FrSky support existed.

hamishwillee, 2015-09-29:
@lvale How about [this addition](http://planner.ardupilot.com/wiki/common-frsky-telemetry/#apm_mavlink_to_frsky_smartport_converter_mavlink_frskysport) for the project? Can you tell me how this project is licensed?

lvale, 2015-09-29:
I'll have an definitive answer for you in a bit, we're still discussing details around GPL3 :).

btw: thanks @hamishwillee looks good :)

hamishwillee, 2015-09-29:
Thanks Luis. I'll keep this open until you have an answer.

lvale, 2015-09-30:
btw the telemetry is not displayed on the receiver but on the transmitter :)

The open source MavLink_FrSkySPort project uses the Teensy USB Development board to convert MAVLink messages to FrSkySmartPort format so that ArduPilot telemetry can be displayed on an FrSky receiver.

hamishwillee, 2015-09-30:
Thanks Luis - fixed that dumb mistake!

hamishwillee, 2015-09-30:
Closing this as section updated and point on license clarified.

closed 2015-09-30 by hamishwillee

ptsneves, 2015-10-26:
Is it possible to add to the Airborne Projects entry that it now supports OpenTX 2.1.3. This is relevant because not many scripts fully work with 2.1.3 OpenTX. If you want me to propose a text I will comply.

hamishwillee, 2015-10-26:
@ptsneves It is better for my maintenance for most of the key selling points to be on your site - so we don't need to keep them updated as they become less relevant or as you change versions.

That said, I am happy to try this - particularly as we can add versioning that will help users determine how relevant the information is. Is there a particular "version" of the converter to which this feature applies?

I was thinking of adding a note [just before the end of this section](https://www.airborneprojects.com/product/apm-mavlink-to-frsky-smartport-converter/) (before "For more information see the") simply saying:

> From version XXX, the converter now supports OpenTX 2.1.3

In addition, the link to your site does not take you to a useful page about the converter: https://www.airborneprojects.com/product/apm-mavlink-to-frsky-smartport-converter/ - is there a better link?

ptsneves, 2015-10-27:
@hamishwillee: Regarding the link, it was inactive just yesterday because I upgraded the server and one of the plugins didn't come back up when we rebooted.

Regarding the new version, we here agreed that so as to not bother the administration of the page and the images, we changed one of them to have a bar regarding the support of 2.1.3.

So my suggestion is to not do anything for now. Is the image compliant with your standards?

hamishwillee, 2015-10-27:
@ptsneves Thanks very much. As you say, link now works, and I think the banner on the product page works well (and really it makes sense to have it here because people could come to that page from anywhere).

> Is the image compliant with your standards?

We don't have any standards on what you do on your site :-) Looks good to me though.

Agree to plan to not do anything on the wiki at this point.