ArduPilot/ardupilot_wiki #128: Dronekit web site broken link to windows installer

HuguesD, 2015-08-14:
Hi,
The link to the windows dronekit api installer on this page: http://python.dronekit.io/guide/getting_started.html, 
is broken, making it impossible to install dronekit on Windows.
Can Hamish fix this (i'm sure he can),
Thx,
Hugues

hamishwillee, 2015-08-14:
Yes, it got broken today, but I thought refixed. If you need it urgently try http://dronekit-assets.s3.amazonaws.com/installers/dronekit-windows-latest.exe

HuguesD, 2015-08-14:
Thx this link works.

Now , next issue with “getting started in windows” is an execution error in the command : api start  “C:\Program Files (x86)\MAVProxy\examples\vehicle_state\vehicle_state.py"

It returns the following error:

STABILIZE> api start "C:\Program Files (x86)\MAVProxy\examples\vehicle_state\vehicle_state.py"

STABILIZE> Exception in APIThread-0: [Errno 2] No such file or directory: '"C:\\Program'

Traceback (most recent call last):

  File "C:\PROGRA~2\MAVProxy\droneapi\module\api.py", line 333, in run

STABILIZE>    self.fn()

  File "C:\PROGRA~2\MAVProxy\droneapi\module\api.py", line 625, in <lambda>

    APIThread(self, lambda: execfile(args[1], g), args[1])

IOError: [Errno 2] No such file or directory: '"C:\\Program'

 

Mavproxy or dronekit does not interpret the directory  correctly. Tried everything I could with other path syntaxes without success.

Any idea how to correct this ?

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

rmackay9, 2015-08-14:
So I think we've moved on from the broken link now so this discussion should also probably move off this wiki issue.

I think MAVProxy doesn't work very well on Windows yet.  It's a bit green.
I'm sure it's not what you want to hear but I always run it using a VM.  http://dev.ardupilot.com/wiki/simulation-2/sitl-simulator-software-in-the-loop/setting-up-sitl-on-windows/

In case it helps, there's drones-discuss and gitter for more detailed help.
https://gitter.im/dronekit/dronekit-python

Also for mavproxy bug reports you can report here:
https://github.com/Dronecode/MAVProxy/issues

hamishwillee, 2015-08-14:
I've moved the discussion to the linked issue. In the meantime the easiest thing to do is to just copy the examples to somewhere else (or get them from git) which doesn't have a space in the path name.

hamishwillee, 2015-08-14:
Also, this is place for wiki issues, not DK - see above links :-)

hamishwillee, 2015-08-14:
@HuguesD I've also patched the broken link - waiting on https://github.com/dronekit/dronekit-python/pull/293 to go in.

HuguesD, 2015-08-15:
Thx a lot Hamish, you're incredibly reactive !
I temporarily moved all python examples in the same directory than mavproxy.exe and it works.
Another correction to do on the getting started page (http://python.dronekit.io/guide/getting_started.html) : in the "Starting MAVProxy" section, "Connecting at startup" sub section, for connection type "Windows computer connected to the vehicle via USB",the "MAVproxy command" should be changed from "mavproxy.py --master=/dev/cu.usbmodem1" to "mavproxy.exe --master=comx" (where x is the com port used to communicate with the vehicle)
Thx

hamishwillee, 2015-08-17:
@HuguesD You're welcome. The Windows USB issue will be fixed as part of https://github.com/dronekit/dronekit-python/pull/296

Closing this, as issues addressed. Remember that correct place to post DK docs issues is here: https://github.com/dronekit/dronekit-python/issues

closed 2015-08-17 by hamishwillee